libretro/Lakka-LibreELEC #962: Lakka not working on Rpi4b 8gb ram version

luxor37, 2020-06-06:
Lakka 2.3.2
[Raspberry pi 4b 8gb ram](https://www.buyapi.ca/product/raspberry-pi-4-model-b-8gb/)

I recently bought the new raspberry pi 4b with 8gn of ram and I tried booting Lakka, but it wouldn't work and instead, I got the following screen (sorry for the bad 4:3, boot option didn't work either):
[image]

After googling a lot and trying different things, I put my sd card into another raspberry pi 4b, but the 1gb ram version and it worked right away. Any way to fix this?

c2d628ba98, 2020-06-09:
I've got the same issue on RPi 4 with 8GB of RAM. I believe this may need an upgrade from the upstream - https://github.com/LibreELEC/LibreELEC.tv/commit/81c2e341552742a5cfa93b6fd082c94fcd429b05

Ntemis, 2020-06-09:
There are more missing pieces but we will make it happen soon, just some patience guys.
We will get there :)
Thank you

natinusala, 2020-06-09:
The LE9 update will probably solve that issue, depending on how recent it is

Ntemis, 2020-06-09:
Good news then @c2d628ba98 & @luxor37 can test the image and confirm

s-sebastian, 2020-06-09:
I had a similar issue and none of the images listed below worked for me:

https://nightly.builds.lakka.tv/2020-02-13/RPi4.arm/Lakka-RPi4.arm-2.9-devel-20200213010100-db102f6.img.gz
https://nightly.builds.lakka.tv/Lakka-LE9.2/latest/RPi4.arm/Lakka-RPi4.arm-2.9-devel-20190917222241-de642b3.img.gz 

I'm not sure if these are the latest available for RPi 4?

I was able to build a working image after merging `LibreELEC 9.2.3` into `Lakka-LE9.2` using docker:

- merge the changes from the upstream

```
$ git clone https://github.com/libretro/Lakka-LibreELEC.git
$ cd Lakka-LibreELEC
$ git remote add upstream https://github.com/LibreELEC/LibreELEC.tv.git
$ git fetch upstream
$ cp Dockerfile /tmp/
$ git checkout Lakka-LE9.2
$ git checkout -b issue-962
$ git merge 9.2.3
$ cp /tmp/Dockerfile .
```

- build and start the container:

```
$ docker build -t lakka .
$ docker run --name lakka --rm -it -v $(pwd):/home/ubuntu lakka
```

- run the build inside the docker container:

```
$ sudo echo "Set disable_coredump false" >> /etc/sudo.conf # https://github.com/sudo-project/sudo/issues/42
$ sudo -R ubuntu .
$ DISTRO=Lakka PROJECT=RPi DEVICE=RPi4 ARCH=arm make image
```

I've also uploaded the files into Dropbox if someone would like to use it before the official build is ready:
https://www.dropbox.com/sh/jm6i41cf9f1yge2/AAAjndcDcSZ3WLgBlLWYd8K_a?dl=0

Ntemis, 2020-06-09:
thank you for the test. so you confirm that this pr fixes this issue? https://github.com/libretro/Lakka-LibreELEC/pull/965

s-sebastian, 2020-06-10:
I didn't test the branch from this PR I'm afraid and I merged [9.2.3](LibreELEC/LibreELEC.tv/tree/9.2.3) into [Lakka-LE9.2](libretro/Lakka-LibreELEC/tree/Lakka-LE9.2) instead.

If you can build and add the images to the nightly builds I can test it.

oldschoolskunk, 2020-06-12:
Same issue.  Can I use NOOBS to boot or 
should I just be patient?

migueldebadia, 2020-06-28:
Hi Sebastian
Thank you very much for your work. Regarding this work, I have a question.
I can't modify the smb.conf file, the terminal shows me a read-only file error, can you help me?
In fact the whole "etc" folder is read-only and I don't know how to change it, I have tried CHMOD and it doesn't work.

svnty, 2020-07-20:
Nightly build 2020-07-19 resolved this issue for me, Lakka now boots on Pi4 8gb model

https://nightly.builds.lakka.tv/2020-07-19/RPi4.arm/

Ntemis, 2020-07-20:
@svnty thank you for the confirmation.
@luxor37 am closing this down.

ggivnish, 2020-09-02:
> Nightly build 2020-07-19 resolved this issue for me, Lakka now boots on Pi4 8gb model
> 
> https://nightly.builds.lakka.tv/2020-07-19/RPi4.arm/

Hi I am unable to access the nightly build from 7-19 which fixed lakka not working on rpi4 8gb.  Does anyone have another link they can share for the noobs.tar file?    Any help would be greatly appreciated

blackmoa, 2020-09-13:
I will put up the newest link in the nightly build
I installed this and was able to start

https://nightly.builds.lakka.tv/2020-09-13/RPi4.arm/

ggivnish, 2020-09-13:
That worked right away, awesome and thanks!!

febs, 2020-12-15:
The issue should be kept open until the official downloadable image for the Rpi4 won't be updated with one which actually works.

Ntemis, 2020-12-15:
Done

febs, 2020-12-15:
I just tried and that ain't working still:
[image: image.png]

tralph3, 2021-01-22:
Can confirm that the version of Lakka for the Pi 4 that's up for download on the website doesn't work, the Pi doesn't boot complaining that "this board requires newer software". Also, the nightly builds work, but have issues (slowdowns on SNES and N64 where there shouldn't be).

apolischuk, 2021-02-25:
Same story, Image downloaded from the website main page isn't working for Rpi4b 8gb. Is it incorrect build or still not ready?

Ntemis, 2021-02-25:
https://nightly.builds.lakka.tv/

ToKe79, 2021-06-10:
"the website" (lakka.tv / le.builds.lakka.tv) had (in early 2021) old build (2.3.x), now new release (3.x) is available to download. this should be not an issue anymore.

closed 2021-06-10 by ToKe79