libretro/Lakka-LibreELEC #964: Download index task failing when updating Lakka on Cubieboard

paulohleal, 2020-06-08:
- Which version of Lakka are you using?
2.3.1
- What system hardware are you using?
Cubieboard
- What did you do?
Main Menu -> Online Updater -> Update Lakka
- What did you expect to happen?
The system update
- What happened instead?
Got the message: "Downloadin index:  Task Failed"

I also checked on the internet about update the config file "/storage/.config/retroarch/retroarch.cfg" but the "core_updater_buildbot_url" is correct.
I also used ping on the host "buildbot.libretro.com" and it toutched the url.
I checked the logs for errors but it did not show any.

yizzy, 2020-06-20:
I have this same issue.
I am running the same verison of Lakka.
However, my hardware is the CUBOX i4Pro.

paulohleal, 2020-06-20:
@yizzy I haven't try.
But I also haven't any luck identifying the CPS ROM's. This is why I am trying to update the system.

yizzy, 2020-06-22:
I see.  I have no luck running Dreamcast ROM's.
Much less luck just detecting Neo Geo ROM's.

closed 2020-08-18 by Ntemis