libretro/Lakka-LibreELEC #968: Error compile RPI

ezar, 2020-06-15:
```
INSTALL      prosystem (target)
[237/270] [DONE] install prosystem
ERROR: /home/ezar/Lakka-LibreELEC/sources/sameboy/sameboy-c9e547c: failed to determine git HEAD
ERROR: /home/ezar/Lakka-LibreELEC/sources/sameboy/sameboy-c9e547c: failed to determine git HEAD
GET      sameboy (git)
    GIT PULL      sameboy
Already up to date.
There is no commit 'c9e547c' on branch 'buildbot' of package 'sameboy'! Aborting!
FAILURE: scripts/install sameboy has failed!
[238/270] [FAIL] install sameboy

The following logs for this failure are available:
  stdout: /home/ezar/Lakka-LibreELEC/build.Lakka-RPi.arm-2.9-devel/.threads/logs/248/stdout
  stderr: /home/ezar/Lakka-LibreELEC/build.Lakka-RPi.arm-2.9-devel/.threads/logs/248/stderr

parallel: This job failed:
package_worker 6 248 270 'install sameboy'
Parallel build failure - see log for details. Time of failure: Mon Jun 15 10:59:42 CEST 2020
make: *** [Makefile:12: image] Error 1
```

by-gnome, 2020-06-15:
https://github.com/libretro/Lakka-LibreELEC/pull/963#discussion_r437198155

ezar, 2020-06-15:
Thnaks!
Changed to PKG_VERSION="44d8b16"

closed 2020-06-21 by ezar