libretro/Lakka-LibreELEC #969: update fails

clevenger23, 2020-06-21:
- Which version of Lakka are you using?
2.3.2
- What system hardware are you using?
Pi 4
- What did you do?
Tried online updater and SSH terminal to update VIA "lakka-update"
- What did you expect to happen?
lakka to download and update
- What happened instead?
Lakka:~ # lakka-update
wget: error getting response: Connection reset by peer
:: Downloading upgrade
Connecting to le.builds.lakka.tv (195.201.56.250:80)
Connecting to le-builds.lakka.tv (104.24.109.116:443)
wget: error getting response: Connection reset by peer

kiicia, 2020-06-28:
it's caused by bug in busybox'es wget implementation (no http->https redirection), see details in [issue 937](https://github.com/libretro/Lakka-LibreELEC/issues/937)

clevenger23, 2020-06-29:
> it's caused by bug in busybox'es wget implementation (no http->https redirection), see details in [issue 937](https://github.com/libretro/Lakka-LibreELEC/issues/937)

15 July 2018 -- BusyBox 1.29.1 (stable)
BusyBox 1.29.1. (git)

Bug fix release. 1.29.1 has fixes for wget (http->https redirect) and sendmail (angle bracket parsing).

TheLastProject, 2020-06-29:
It's definitely not because of the http to https redirect.

If you copy /usr/bin/lakka-update to your home directory and change http to https in the url it will still fail.

ToKe79, 2020-07-19:
I have removed the redirect from http to https, can you try now?

Ntemis, 2020-08-09:
no feedback on this closing down

closed 2020-08-09 by Ntemis