libretro/Lakka-LibreELEC #979: Graphical issues on Eee 1005 HA

Luke1002, 2020-07-13:
Hi everyone, I recently found an old laptop in my house and I tried to install Lakka for some simple retro gaming.
The laptop is an Asus Eee 1005HA ([here's the specs](https://www.asus.com/Laptops/Eee_PC_1005HA_Seashell/specifications/)), so there should not be any problem with the 32-bit version for generic pc. However when I try to boot Lakka (whether in live mode or from the hard disk), after the logo this [glitched screen](https://imgur.com/xNYCeFk) appears.
I tried the usb drive on another computer and live mode worked perfectly.

As for the logs, I really don't understand how I am supposed to find them, since the methods listed on the website seem to require for the user to get the os in a usable state.

gouchi, 2020-07-13:
Hi,

May you try to provide [graphic card log](http://www.lakka.tv/doc/Troubleshooting-Lakka/#graphic-card-logs) ? 

If it is not [Intel GMA 500](https://en.wikipedia.org/wiki/Intel_GMA#PowerVR_SGX_535_based), please try to edit `menu_shader_pipeline = "0" `in [retroarch.cfg](http://www.lakka.tv/doc/Configuring-Lakka). If it is still not working, try to use rgui menu driver with `menu_driver = "rgui"`

Thank you for your tests.

Luke1002, 2020-07-14:
Graphic card log

```
00:02.0 VGA compatible controller [0300]: Intel Corporation Mobile 945GSE Express Integrated Graphics Controller [8086:27ae] (rev 03)
        Subsystem: ASUSTeK Computer Inc. Mobile 945GSE Express Integrated Graphics Controller [1043:8340]
        Kernel driver in use: i915
00:02.1 Display controller [0380]: Intel Corporation Mobile 945GM/GMS/GME, 943/940GML Express Integrated Graphics Controller [8086:27a6] (rev 03)
```

This is the general log.
```
[INFO] RetroArch 1.7.8 (Git ea7e682253)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Atom(TM) CPU N280   @ 1.66GHz
[INFO] Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
[INFO] Built: Sep 16 2019
[INFO] Version: 1.7.8
[INFO] Git: ea7e682253
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 1 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Mode 0: (1024x600) 1024 x 600, 60 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1024x600.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
[INFO] [EGL]: Current context: 0xb4308e38.
[INFO] [KMS]: New FB: 1024x600 (stride: 4096).
[INFO] [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 945GME x86/MMX/SSE2.
[INFO] [GL]: Version: 1.4 Mesa 17.1.3.
[INFO] [GL]: Using resolution 1024x600
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[INFO] [GLSL]: Checking GLSL shader support ...
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] Setting up menu pipeline shaders for XMB ... 
[INFO] [GLSL]: Compiling ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling simple ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling snow shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[... 35 lines cut ...]
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Menu]: Found menu display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
function is no-op
[INFO] [LED]: LED driver = 'null' 0x8653594
[INFO] [MIDI]: Initializing ...
ALSA lib /home/natinusala/release/Lakka-LibreELEC/build.Lakka-Generic.i386/alsa-lib-1.1.9/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed (list of input devices unavailable).
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].
function is no-op
function is no-op
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [KMS]: New FB: 1024x600 (stride: 4096).
function is no-op
[INFO] [KMS]: New FB: 1024x600 (stride: 4096).
```

Luke1002, 2020-07-14:
> Hi,
>
> May you try to provide [graphic card log](http://www.lakka.tv/doc/Troubleshooting-Lakka/#graphic-card-logs) ?
>
> If it is not [Intel GMA [...]

I changed the two values in retroarch.cfg accordingly, but nothing different happens.

gouchi, 2020-07-16:
Thank you for the log, indeed it seems there is no fatal issue.

You did stop retroarch before editing the configuration file ?

May you try to edit retroarch.cfg with `video_monitor_index = "1"` ?

Luke1002, 2020-07-17:
> Thank you for the log, indeed it seems there is no fatal issue.
> 
> You did stop retroarch before editing the configuration file ?
> 
> May you try to edit retroarch.cfg with `video_monitor_index = "1"` ?

It seems like I didn't save the config file correctly, so I tried the settings previously suggested alongside this new one and live mode indeed started up correctly. Being extremely happy for the success, I immediately tried installing Lakka on the laptop and it obviously failed. Is there a way to modify the config file in such a way that lets me install the os with the settings already changed?

P.S. Sorry for my English, please feel free to ask for any further explanation if it's not clear.

gouchi, 2020-07-18:
If you installed it you should to the same. 

So you should be able to edit [cmdline](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc) and enabled ssh (just add `ssh` to cmdline) to edit the configuration file as you did before by ssh to your station.

Otherwise, you can boot with another live linux distro to edit directly the configuration on the disk.

Luke1002, 2020-07-24:
I tried a live distro but the retroarch folder is blocked. I am not too familiar with Linux, but it seems like I need administrator privileges. I also tried _sudo nano /storage/.config/retroarch/retroarch.cfg_ but it seems like the folder doesn't exist.

gouchi, 2020-07-28:
Depending of the distro you are using but you have a little explanation [how to access your hard drive from live cd](https://askubuntu.com/questions/190239/how-to-access-hard-drive-files-from-livecd). Moreover, if you don't see the directory try to use `CTRL+h `to display hidden files.

Ntemis, 2020-08-09:
@Luke1002 try the nightlies builds please, bye

closed 2020-08-09 by Ntemis