libretro/Lakka-LibreELEC #989: Syslinux error on booting from a chromebook

gogilberto, 2020-07-20:
I posted this earlier on the forums, but i thought that it might fit better here

To start off, I am using the latest stable version of Lakka which I installed onto a MicroSD card using the chrome recovery tool.

The current Chromebook I am using is the HP Chromebook 14 G3 iirc. I used Mr. Chromebox’s script to update SeaBios on my Chromebook.

When i use SeaBios to boot from my MicroSD card, syslinux 6.04-pre1 appears. 

Here is where i encounter a problem, Syslinux starts up and it says that it is waiting for boot or press TAB for other options (Something like this) and then in the next line it just says “boot:” And that is as far as i got, i waited 10 minutes to see if this was just loading or something but sure enough, nothing came of it. I saw no Lakka symbol nor did i see a Lakka menu.

Any help on getting Lakka to run would be greatly appreciated.

ToKe79, 2020-07-20:
Have you tried typing `live` and hitting the enter key?

gogilberto, 2020-07-20:
Damn i feel stupid now, that was totally a step that i missed out on when reading about Lakka, thats my bad, ignore this issue and move along

closed 2020-07-20 by gogilberto