libretro/Lakka-LibreELEC #994: Build fails with GCC  10

mellowcandle, 2020-07-25:
- Which version of Lakka are you using?
Latest master
- What system hardware are you using?
Manjaro Latest release, x86.
- What did you do?
Build with: 
DISTRO=Lakka PROJECT=Qualcomm DEVICE=Dragonboard ARCH=arm make image

/home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/toolchain/bin/host-gcc -DHAVE_CONFIG_H -I. -I/home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/binutils-2.32/binutils  -I. -I/home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/binutils-2.32/binutils -I../bfd -I/home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/binutils-2.32/binutils/../bfd -I/home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/binutils-2.32/binutils/../include -DLOCALEDIR="\"/home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/toolchain/share/locale\"" -Dbin_dummy_emulation=bin_vanilla_emulation  -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wstack-usage=262144 -I/home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/binutils-2.32/binutils/../zlib -g -O2 -MT prdbg.o -MD -MP -MF .deps/prdbg.Tpo -c -o prdbg.o /home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/binutils-2.32/binutils/prdbg.c
In file included from /home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/binutils-2.32/gold/debug.h:29,
                 from /home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/binutils-2.32/gold/fileread.cc:42:
/home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/binutils-2.32/gold/errors.h:87:50: error: 'string' in namespace 'std' does not name a type
   87 |   undefined_symbol(const Symbol* sym, const std::string& location);
      |                                                  ^~~~~~
/home/stdcall/dev/Arcade/Lakka-LibreELEC/build.Lakka-Dragonboard.arm-3.0-devel/binutils-2.32/gold/errors.h:29:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
   28 | #include "gold-threads.h"
  +++ |+#include <string>
   29 | 
make[5]: *** [Makefile:1133: fileread.o] Error 1

Ntemis, 2020-07-25:
Build system is based on debian Ubuntu. Archlinux is not supported

closed 2020-07-25 by Ntemis

mellowcandle, 2020-07-25:
That's just a heads up, eventually Ubuntu will transition to GCC 10.

sierrahotel, 2021-07-15:
For those finding this here's a possible workaround:
You can rollback to gcc 9.2.0, this worked for me.
This will require the removal of sysprof and any of it's dependencies however.
As a minimum you'll need to remove package gnome builder (a sysprof dependency), then simultaneously downgrade gcc and gcc-libs to 9.2.0 answering y when asked to remove sysprof.