libretro/Lakka-LibreELEC #999: GCC 8.3.0 warning '-mcpu conflicts with -march'

by-gnome, 2020-08-03:
The issue for branch master.

When glibc, keyutils, elfutils (and maybe something else) are built for Cortex-A9, the gcc-8.3.0 produces a warning:
`switch -mcpu=cortex-a9 conflicts with -march=armv7-a switch`
which is treated as an error:
`cc1: error: switch -mcpu=cortex-a9 conflicts with -march=armv7-a switch [-Werror]`

Flags -mcpu and -march are defined in config/arch.arm:
```
TARGET_SUBARCH=armv7-a
TARGET_EXTRA_FLAGS="-mcpu=$TARGET_CPU"
TARGET_VARIANT="${TARGET_SUBARCH}${TARGET_CPU_FLAGS}"
TARGET_CFLAGS="-march=$TARGET_VARIANT -mtune=$TARGET_CPU -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated $TARGET_EXTRA_FLAGS"
```

Maybe this issue can be solved by removing one of them?

by-gnome, 2020-08-07:
For ARM we have to define -march with -mtune or -mcpu.

Ntemis, 2020-08-07:
I will investigate thak you for the report

Ntemis, 2020-08-07:
can you check now?

by-gnome, 2020-08-07:
Still the same.
I use https://github.com/by-gnome/Lakka-LibreELEC/commit/5fe534980a8f9777922f2afb741b48f92b6c9eb8 to build.

Ntemis, 2020-08-07:
hmm i told you to check after i updated gcc to 9.3 but i saw you didn't pull the changes

Ntemis, 2020-08-07:
please pull test and close issue if ok. thanks

by-gnome, 2020-08-07:
I was building this branch https://github.com/by-gnome/Lakka-LibreELEC/commits/master-wandboard

Ntemis, 2020-08-07:
please pull master before build. Also welldone that merge looks clean i think i can merge on master and then you can close the old one if interested

by-gnome, 2020-08-07:
Solved by commit https://github.com/libretro/Lakka-LibreELEC/commit/73778f9061e1ab77b7a08c9c5c0581bbafce7954, thnx.

closed 2020-08-07 by by-gnome