libretro/Lakka-LibreELEC #1006: Add Support for XOW driver on LAKKA

joaodforce, 2020-08-09:
There is this new Driver for Xbox One Wireless Adapter for Linux

called "XOW" https://github.com/medusalix/xow

I've tested in some platforms already and it works very well, I've tried installing it on Lakka on a Raspberry PI 3.

But I could not because of the closed nature of the release builds.

Ntemis, 2020-08-09:
do you really want us shutdown? why?
By using xow, you accept Microsoft's license terms for their driver package
No thank you but NO, bye

closed 2020-08-09 by Ntemis

joaodforce, 2020-08-09:
A simple explanation would suffice, there is no need to be so condescending.
Thanks for the clarification anyway.

Ntemis, 2020-08-09:
Abolutely no condescendation included here, you come here asking for this before studying the driver at all, something that could have a malicious consequence to Lakka if included. Yes i don't take things like these lightly but am i really the one at fault here?
Sorry if i hurt your feelings but maybe in the future you are more careful about asking something before you investigate it further.
Greetings and thank you for trying to make lakka distro better as we do appreciate your contribution  and time taken to report this driver.