libretro/Lakka-LibreELEC #1007: sync issues when using rewind on pie 4

Cibomatto2002, 2020-08-10:
I'm using a nightly build 2020-07-24 it also did this on the 2.3.2 build but it seems worse on the 2.3.2 build. I'm using a 2GB raspberry pi 4. It does not happen all the time just sometimes.

I noticed it can happen on a bios screen to without using the rewind on the mgba core this may happen on other cores to but i'm only using the mgba core right now.

natinusala, 2020-08-10:
What is the problem you are encountering exactly? Does it happen on Raspbian too?

Cibomatto2002, 2020-08-10:
I get screen tearing on rewind sometimes.

natinusala, 2020-08-10:
Does it continue after rewinding is done?

Cibomatto2002, 2020-08-10:
It does it right after rewinding is done. 

Also is the save directory bug going to get fixed before Lakka 3.0 comes out?

Ntemis, 2020-08-20:
@Cibomatto2002 can you try https://nightly.builds.lakka.tv/2020-08-20/RPi4.arm/Lakka-RPi4.arm-3.0-devel-20200820023653-59d5c4b.img.gz please

Cibomatto2002, 2020-08-21:
I have tried this for 30 minutes so far so good. :)