libretro/Lakka-LibreELEC #1011: LAKKA on Pi 4 B 8gb no boot, no start, send error message on booting action

palitofat, 2020-08-18:
If you are sure that it's a bug in Lakka, please answer the following questions:

- Which version of Lakka are you using?
Lasted and previus vers
[image]
ions is same!
**Lakka-RPi4.arm-3.0-devel-20200817005022-50aca1f.img.gz**

- What system hardware are you **using?**
**Raspberry pi 4 B 8 giga bites**

- What did you do?
Burn on sd card of 32gb class10 with balena

- What did you expect to happen?
I expect run LAKKA OS but no funtion

- What happened instead?
To moment to chance sd for lakka sd. Generate a n error boot secuense.

palitofat, 2020-08-18:
Apreciated help me with this unconvenient please. 
Sorry with my English

Ntemis, 2020-08-18:
you haved to start lakka with lakka sd in and system power off then on, you cant change sd on the fly if i get it right on what you are reporting here "To moment to chance sd for lakka sd. Generate a n error boot sequence."

Ntemis, 2020-08-18:
If you are trying to boot from usb then try an sdcard and report back

palitofat, 2020-08-18:
Obviusly I power off raspberry pi4. Then I extract sd os. And put on the new sd lakka. And power on. Inmediatly put the message cant boot

palitofat, 2020-08-18:
No I not put sd lakka os on port usb 3.5. I put lakka on port sd for os boot. With power off pi4

Ntemis, 2020-08-18:
Is clear now,thanks for the report, it will be investigated asap

palitofat, 2020-08-19:
one more thing, once the sd is burned with lakka. I cannot visualize what is inside the sd. I cant see the boot files, config, kernel, etc.
I think the lakka operating system for pi 4 8gigas is corrupt.

palitofat, 2020-08-20:
Good morning, any solution o no yet

Ntemis, 2020-08-20:
I will link you to the latest rpi4 image to test 
https://nightly.builds.lakka.tv/2020-08-20/RPi4.arm/Lakka-RPi4.arm-3.0-devel-20200820023653-59d5c4b.img.gz
Let me know if the issue is fixed, thanks

palitofat, 2020-08-20:
Hello Ntemis
nothing yet.
I tried to burn it with balena etcher and it gave error during the burning process. after I finish this, turn off the rasp, exchange cards and the same problem. I send you photos of the processes. Thank you
[image]
[image]
[image]

Ntemis, 2020-08-20:
Please write the image again with out extracting it and make sure balena etcher verified the burn.

palitofat, 2020-08-20:
[image]
[image]
[image]

Balena configured with your instrucctions 
No expulse
Validate processes activated

Hello again.
Same situation
No correct burn
Look the pictures

[image]

palitofat, 2020-08-20:
Remember my raspberry is 4b and 8 gb of memory ram. I dont nowk if that is a problem. I bougth recently on amazon. 
Maybe bios
Maybe Lakka files
I dont nowk

palitofat, 2020-08-20:
Hello any avance?

Ntemis, 2020-08-20:
The issue is on your part there is nothing wrong with lakka image https://imgur.com/ZzZklYx

palitofat, 2020-08-20:
Ahhh ok

closed 2020-08-20 by Ntemis