libretro/Lakka-LibreELEC #1014: The Great GBA Core Disabling

SethBarberee, 2020-08-26:
As we know, there are a few GBA libretro emulators in Lakka. To list them all:

* mgba
* vba-next
* vbam
* meteor
* gpsp
* meteor
* beetle-gba

I mainly test on the RPi Zero W so here's my angle. I am proposing to only enable **mgba** for GBA and not build the rest. This isn't much work on our part aside from disabling the builds on `distributions/Lakka/options` or even deleting the package.mk for those extra ones. The playlist default is mgba anyway so the user has to have intent to use the other variants when any other ones are built.

My rationale for this is:
1) mgba has the most accurate for GBA emulation.
       It's the recommended emulator for ROM hacks too because of its accuracy and softpatching ability
2) mgba is pretty performant even on low end devices on Lakka
        I primarily test Pokemon, Advance Wars, and Mario Kart on my RPi 0 W and haven't really experienced noticeable slowdown
3) Building less cores saves space
         Less cores means more space for roms (and because Generic X86_64 builds every emulator in the repo)

I would love to hear other opinions/thoughts about this proposal in the comments.

Ntemis, 2020-08-26:
I see two meteor entries in your list, is it a typo?
Anyway my thoughts are,
Meteor and beetle-gba can go and we decide to keep either vbam or vba-next

gouchi, 2020-08-26:
+1 for mgba as it [the](https://github.com/libretro/ludo/blob/master/Makefile#L4) [one](https://github.com/libretro/ludo/wiki#emulated-consoles) used in Ludo ;)

PR #1015 (2020-08-27): MAME/GBA cleanup
As discussed in #1014 and on discord. 

Fixes #1014 

* vba-next/meteor/beetle-gba are removed
* vba-m/Genesis Plus GX is pointed to build from the upstream repo
* mame2003 and mame2003-midway have been deleted
* mame2016 has been disabled bc of build issues on H3 so pointed playlist entry to normal MAME
      Was mentioned on discord to change default MAME to mame2003-plus except for Generic so did that
closed 2020-08-27 by Ntemis