libretro/Lakka-LibreELEC #1016: Segfault on Nightly (Aug 26-28, 2020) from launching Arcade/NES/SNES/Atari 2600 cores

mcbeth, 2020-08-28:
I decided to try the nightlies on my RPi4 2GB to see if they fixed my problems with DOSBOX my controller.  It did, but now any cores I've tried (for Arcade, NES, SNES, Atari 2600) crash with a segmentation fault if run from the playlists.  They work fine if I "Load Content" and then pick the core.  ScummVM and DOSBOX appears to work fine.

I finished building a debug version from git last night, but the backtrace is corrupt when run under gdb
gdb retroarch
gdb> run -v
....
[INFO] [KMS]: New FB: 3840x2160 (stride: 15360).
[INFO] [KMS]: New FB: 3840x2160 (stride: 15360).
Thread 1 "retroarch" received signal SIGSEGV, Segmentation fault.
gdb> bt
#0  __GI___libc_free (mem=0x2) at malloc.c:3109
#1  0xb54a9b24 in __GI___libc_realloc (oldmem=0x2, bytes=0) at malloc.c:3152
#2  0x00053458 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

The same segment of log when it successfully runs from "Load Content" is:
[INFO] [KMS]: New FB: 3840x2160 (stride: 15360).
[INFO] [KMS]: New FB: 3840x2160 (stride: 15360).
[INFO] [Environ]: SET_SUBSYSTEM_INFO.
[INFO] [CONTENT LOAD]: Updating firmware status for: /tmp/cores/fbneo_libretro.so on /tmp/system
...

SethBarberee, 2020-08-28:
Which cores are you using for each of these systems?

mcbeth, 2020-08-28:
Every core installed by default that I've tried on the RPi4 has this behavior (except DosBOX_SVN and ScummVM).  FinalBurnPro Neo, MAME 2003, MAME 2010, MAME 2015, Stella 2014, FCEUmm, Mesen, Nestopia UE, bsnes, Snes9x - Current, PCSX  ReARMed.

Looking at the backtraces above (and the fact the games work from Load Content with the same cores, but not from the playlists/thumbnail thingie), I honestly don't think it's getting to any of the cores, and the crash is in retroarch somewhere.

Ntemis, 2020-08-28:
Correct, retroarch is trying to load @3840x2160 Rpi4 gpu drivers are not in the best state to support such high demand

mcbeth, 2020-08-28:
Okay, 2.3.2 ran them fine, nightlies through Load Content runs them fine.  Playlists don't.  I'll figure out how to force the resolution down lower and see if that helps.

Thank you for your time and consideration

SethBarberee, 2020-08-28:
> Okay, 2.3.2 ran them fine, nightlies through Load Content runs them fine. Playlists don't. I'll figure out how to force the resolution down lower and see if that helps.

You can edit the `config.txt` with your desired resolution. 
https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

mcbeth, 2020-08-29:
Okay, I've tried all the HDMI 16:9 modes down to 480p, and the crashing behavior is identical to when I let it follow the EDID from the monitory (modes change, segmentation fault right after reporting the New FB, unless I go the Load Content route, when it works.  Thanks for the suggestions, if  there's anything else I can do to help, I'm happy to try

Ntemis, 2020-08-31:
I will have something ready soon for rpi4 for you to test.

mcbeth, 2020-09-02:
The Sept 1st nightly appears to fix my report

Ntemis, 2020-09-02:
you want this closed?

mcbeth, 2020-09-08:
I believe that my complaint is fixed, so from my perspective, I have no concerns if it is.  Thank you for your hard work.

closed 2020-09-08 by mcbeth