libretro/Lakka-LibreELEC #1018: Getting screen tearing in the 2003 Plus core

Cibomatto2002, 2020-08-31:
using the nightly build 8-31-2020

Pie 4

I have tried a few games in MAME 2003 Plus and you can see screen tearing.

Play the game Alpha Mission II / ASO II - Last Guardian you can let the demo run to see it.

Ntemis, 2020-08-31:
Thank you for the report. An rpi4 is coming my way.
I will have a shining new image for you to test asap i have it

Cibomatto2002, 2020-08-31:
I just downloaded the new nightly September 1 tearing still there.

Cibomatto2002, 2020-09-04:
Played some nes today and seen some tearing played it for over a hour.

Ntemis, 2020-09-10:
@Cibomatto2002 please test this image https://nightly.builds.lakka.tv/2020-09-10/RPi4.arm/Lakka-RPi4.arm-3.0-devel-20200910021956-82f7874.img.gz

Cibomatto2002, 2020-09-11:
Still there I made a video for you to see maybe it can help.

https://youtu.be/qEy1_abb6jE

Ntemis, 2020-09-11:
the "tearing" you see can be core related or even how your screen is handling the very fast refresh rate but i wouldn't blame lakka for this either we can fix such issues.
Please report the "tearing" you see to the core maintainer for the individual core you have it.
Am closing this down. Thank for the video btw, it was enlightening.

closed 2020-09-11 by Ntemis

Cibomatto2002, 2020-09-11:
I hope I did not waste your time working on this I felt like I was that's why I made a video.

Cibomatto2002, 2020-09-11:
I just wanted to tell you I got it working disabling the bypass audio skew fixed it.