libretro/Lakka-LibreELEC #1023: GPIO Controller raspberry pi

Turro75, 2020-09-07:
Hi All,

I created a quick wiki page which explains how to set a generic 2 players controller by using GPIO. it is based on game-hat driver.
Since I didn't find anything in the web I think it could be useful sharing it to other users.

[https://github.com/Turro75/Lakka-LibreELEC/wiki/Raspberry-pi-and-GPIO-controllers-with-Lakka-2.3.2](url)

Best Regards

tavuntu, 2020-11-16:
Do you think this will work for a different button mapping? I'm talking about the configuration expected by the [SNESDev](https://github.com/petrockblog/SNESDev-RPi) driver (I have [this joystick](https://monsterjoysticks.com/clear-deluxe-arcade-controller-kit-for-raspberry-pi-classic) with a hat that covers like 2/3 of the GPIO pins, and this is just for 1 controller).

So yes, in short, I'm trying to compile Lakka along with the SNESDriver and have it ready to use, not sure if I explained mysef or if I'm in the right place.

Turro75, 2020-11-26:
Well I have a quick look at this, You can definitely get it working bypassing (throwing away) the GPIO interface PCB and wiring the controls directly to GPIO. 
Otherwise You may ask to Lakka developers to create a package for the snesdev driver

rafaelmartines, 2020-12-30:
Your tutorial works great on Lakka 2.3.2, but unfortunatelly doesn't work on Nightly Builds :\ I'm worried that I'll not be able to upgrade to Lakka 3.0 on future, because my GPIO controllers are not recognized natively or by doing your steps.

ToKe79, 2021-06-10:
Does this work in 3.x? Can we use the write-up in the Lakka wiki / docs (with proper credit, ofc)?

Turro75, 2021-06-13:
https://github.com/Turro75/mk_arcade_joystick_rpi/wiki/Raspberry-pi-and-GPIO-controllers-with-Lakka-3

Not sure the released 3.x includes the driver as default

ToKe79, 2021-06-14:
yes, 3.x release includes the `mk_arcade_joystick_rpi` package. RPi3.aarch64 was added after 3.1 release, initial nightly builds did not include the package, but it is included in nightly builds since (including) June 11 2021.
The question was if the guide (as it is written) works also for 3.x release (or some steps need to be changed) and if we can point to in in our wiki / documentation.

Turro75, 2021-06-14:
Tested Today the official 3.1 on a Rpi4 (aarch64) and a rpi3a+ (32bit), both worked as described in the lakka3 guide.

closed 2021-06-30 by ToKe79