libretro/Lakka-LibreELEC #1024: Mupen64Plus-Next not working on Orange Pi Plus 2E

d-apps, 2020-09-08:
I am trying to play some N64 games but everytime I try to run a game I get a blackscreen and RetroArch restarts. 

Before I install Lakka OS I had RetrOrangePi installed and I was using Mupen64Plus-Next, the core was working fine, almost all of my romset was working, decent/playable fps.

Since Mupen64Plus-Next was not working on Lakka OS I tried ParaLLel N64 core, at least games are booting but the fps is bad, I had some headackes trying to change the GFX Plugin and other settings (requires restart but when I try to restart it freezes and when it doesn't freeze it doesn't save the new settings).  So I tried to change retroarch core cfg file, but even lowering the settings fps were too slow.

- Which version of Lakka are you using?
   **2.3.1**

- What system hardware are you using?
   **Orange Pi Plus 2E**

- What did you do?
  **Select a Nintendo 64 game, select Mupen64Plus-Next, run.**

- What did you expect to happen?
  **Run the game.**

[Mupen64Plus-Next log.txt](https://github.com/libretro/Lakka-LibreELEC/files/5185759/Mupen64Plus-Next.log.txt)

Ntemis, 2020-09-08:
please try this image and let me know
https://nightly.builds.lakka.tv/2020-09-08/H3.arm/Lakka-H3.arm-3.0-devel-20200908000709-04831f7-orangepi-plus2e.img.gz

d-apps, 2020-09-08:
> please try this image and let me know
> https://nightly.builds.lakka.tv/2020-09-08/H3.arm/Lakka-H3.arm-3.0-devel-20200908000709-04831f7-orangepi-plus2e.img.gz

Thanks but with this nighly version WiFi is not working, it didn't find my network after scanning, I couldn't transfer my roms through network so I connected my HD with roms, same behavior for Mupen64Plus-Next, after loading a rom RetroArch reboots and now ParaLLel N64 has the same behavior, before it was at least loading games.

Ntemis, 2020-09-08:
i need some more helpful logs as Segmentation fault doesnt help at all

d-apps, 2020-09-08:
> i need some more helpful logs as Segmentation fault doesnt help at all

Sorry but without WiFi I have no idea how to execute retroarch in verbose mode, I was using SCP and using the Command option to input the command: `systemctl stop retroarch.service ; retroarch -v`.

Could you please guide me to generate logs without WiFi?

d-apps, 2020-09-08:
I opened WiFi option to see if logs might show something about WiFi not working and I tried to run a rom with Mupen64Plus-Next.

[Nightly 3.0 WiFi Mupen64Plus log.txt](https://github.com/libretro/Lakka-LibreELEC/files/5189440/Nightly.3.0.WiFi.Mupen64Plus.log.txt)

Ntemis, 2020-09-08:
RTL8189ETV cant work with mainline kernel it needs https://github.com/jwrdegoede/rtl8189ES_linux 
Maybe on next lakka version based on le 10 branch.
Feel welcome to sent a pr if you still need this included

Ntemis, 2020-09-08:
for the rest of your issues since everything else is in order you can open a new issue here: 
- https://github.com/libretro/mupen64plus-libretro-nx
and here 
- https://github.com/libretro/parallel-n64 
Providing the needed logs you will be asked for 
Since wifi is marked as "will not be fixed" and rest is not lakka related am closing this down

closed 2020-09-08 by Ntemis

d-apps, 2020-09-08:
> RTL8189ETV cant work with mainline kernel it needs https://github.com/jwrdegoede/rtl8189ES_linux
> See https://linux-sunxi.org/Wifi#RTL8189ES_.2F_RTL8189ETV
> Maybe on next lakka version based on le 10 branch.
> Feel welcome to sent a pr if you still need this included

Thank you, but about WiFi, since version 2.3.1 the WiFi is working, shoudn't new version be working too? Or does it need to be implemented again for new versions of Lakka OS?

Ntemis, 2020-09-08:
At that time we were using the official kernel