libretro/Lakka-LibreELEC #1025: Installing libva fails when building master

jordemort, 2020-09-09:
I'm trying to build Lakka from master, and failing because libva can't find libGL:

```
checking GL/gl.h usability... no     
checking GL/gl.h presence... no
checking for GL/gl.h... no    
checking GL/glx.h usability... no
checking GL/glx.h presence... no               
checking for GL/glx.h... no            
checking for glXCreateContext in -lGL... no
configure: error: VA/GLX explicitly enabled, but libGL couldn't be found
FAILURE: scripts/install libva has failed!                                                                             
[217/351] [FAIL] install libva
```
- Which version of Lakka are you using? `1c22a82854c86afd5da489c88b72007d9d2828ea`
- What system hardware are you using? x86_64 Linux box (tried my Debian buster machine and the Ubuntu container defined in the Dockerfile)
- What did you do? `DISTRO=Lakka PROJECT=Generic ARCH=x86_64 make image`
- What did you expect to happen? I expected to build Lakka
- What happened instead? Building failed, with the error re: libva above

Ntemis, 2020-09-09:
I will start a clean build and check.

Ntemis, 2020-09-21:
Found no issues  ¯\_(ツ)_/¯

closed 2020-09-21 by Ntemis

jordemort, 2020-09-21:
@Ntemis Yeah I think https://github.com/libretro/Lakka-LibreELEC/commit/d8cadf4dce42664db5247cb0522ee3ed4e1e9c26 fixed it