libretro/Lakka-LibreELEC #1026: Unable to add smb mount

brunnels, 2020-09-10:
With image built from latest master I am unable to add cifs mounts as documented here http://www.lakka.tv/doc/Serving-ROMs-from-a-NAS/

It appears that mount doesn't support cifs because I don't see it in the output of `cat /proc/filesystems`

The `systemctl status` and `mount -t cifs` error is `no such device`

If I build the latest LibreELEC.tv master image in the exact same manner it works as documented.

I also tried nfs mount and that does not work either.

I am building in a docker container built with `docker build --pull -t lakka tools/docker/xenial`  from the project root

Then from the project root I run 
```
docker run --rm -v `pwd`:/build -w /build -it \
-e MTPROGRESS=yes \
-e AUTOREMOVE=yes \
-e PROJECT=Allwinner \
-e ARCH=arm \
-e DEVICE=H6 \
-e UBOOT_SYSTEM=orangepi-one-plus \
lakka make image
```

Ntemis, 2020-09-10:
Thanks for the report. It checks out cifs support was not enabled in kernel.

gouchi, 2020-09-28:
Indeed [CONFIG_CIFS is not set for Allwinner](https://github.com/libretro/Lakka-LibreELEC/blob/master/projects/Allwinner/linux/linux.arm.conf#L4879).

closed 2020-10-20 by Ntemis