libretro/Lakka-LibreELEC #1027: Generic X64 Nightly Hangs on Lakka Splash Screen

MiniSpoon, 2020-09-10:
Nightly builds newer than 2020-09-07 will not fully boot on my computer. The display hangs at the Lakka berry slash screen. SSH actively rejects connections, so it seems the OS is booted, but the graphics driver isn't loading.

I tried builds from 02 September - 10 September. 02 September and 07 September work. 08 September and later do not work. 

Specs:
Intel Kaby Lake CPU
Integrated HD 630 GPU
B250 Motherboard Chipset

Here is some command line output detailing the linux kernel and video hardware. This was from the 07 September build.

Lakka:~ # uname -a
Linux Lakka 5.4.61 #1 SMP Mon Sep 7 00:46:30 CEST 2020 x86_64 GNU/Linux

Lakka:~ # lspci -nnk | grep -A 3 VGA
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 630 [8086:5912] (rev 04)
        DeviceName:  Onboard IGD
        Subsystem: Micro-Star International Co., Ltd. [MSI] HD Graphics 630 [1462:7a67]
        Kernel driver in use: i915

Lakka:~ # lspci -vvv -d 8086:5912
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 630 (rev 04) (prog-if 00 [VGA controller])
        DeviceName:  Onboard IGD
        Subsystem: Micro-Star International Co., Ltd. [MSI] HD Graphics 630
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 123
        Region 0: Memory at de000000 (64-bit, non-prefetchable) [size=16M]
        Region 2: Memory at c0000000 (64-bit, prefetchable) [size=256M]
        Region 4: I/O ports at f000 [size=64]
        [virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express (v2) Root Complex Integrated Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0
                        ExtTag- RBE+
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis-, LTR-, OBFF Not Supported
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
                         AtomicOpsCtl: ReqEn-
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable- 64bit-
                Address: fee02004  Data: 4022
        Capabilities: [d0] Power Management version 2
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [100 v1] Process Address Space ID (PASID)
                PASIDCap: Exec- Priv-, Max PASID Width: 14
                PASIDCtl: Enable- Exec- Priv-
        Capabilities: [200 v1] Address Translation Service (ATS)
                ATSCap: Invalidate Queue Depth: 00
                ATSCtl: Enable-, Smallest Translation Unit: 00
        Capabilities: [300 v1] Page Request Interface (PRI)
                PRICtl: Enable- Reset-
                PRISta: RF- UPRGI- Stopped+
                Page Request Capacity: 00008000, Page Request Allocation: 00000000
        Kernel driver in use: i915

Please let me know if there is any other information I can provide. Thank you.

Ntemis, 2020-09-10:
Can you try to ssh to it? if you can i will need a dmesg log

MiniSpoon, 2020-09-10:
Is there a way to get SSH working without enabling the service through the GUI? It rejects my connection attempts.

Ntemis, 2020-09-10:
nice point, i think we should enable ssh as default

gouchi, 2020-09-10:
You can boot with `live ssh` for the live mode to enable SSH.

MiniSpoon, 2020-09-11:
I was able to get the log from live ssh.
[https://drive.google.com/file/d/1lY3MNG7nhI-vu2xtAxxJJB5j_4xuuNAl/view?usp=sharing](url)

Ntemis, 2020-09-11:
Access denied

MiniSpoon, 2020-09-11:
Please try this.
[https://dpaste.com/5NNMZW98P](url)

Ntemis, 2020-09-11:
yeap no gpu for you

MiniSpoon, 2020-09-19:
Rolling back the Mesa Iris drivers did the trick. The latest nightly build works (19 Sept 2020).

closed 2020-09-19 by MiniSpoon