libretro/Lakka-LibreELEC #1029: Cannot load GL context for UHD Graphics 605

MrJustreborn, 2020-09-12:
Spec:
INTEL NUC with
Silver J5005
UHD Graphics 605

Lakka (official): devel-20200910004523-82f7874 (Generic.x86_64)

> lspci -nnk | grep -A 3 VGA
00:02.0 VGA compatible controller [0300]: Intel Corporation UHD Graphics 605 [8086:3184] (rev 03)
	DeviceName: Onboard - Video
	Subsystem: Intel Corporation UHD Graphics 605 [8086:2072]
	Kernel driver in use: i915

> systemctl stop retroarch.service ; retroarch -v
[INFO] RetroArch 1.9.0 (Git 5e551dd92b)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Pentium(R) Silver J5005 CPU @ 1.50GHz
[INFO] Capabilities:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES
[INFO] Built: Sep 10 2020
[INFO] Version: 1.9.0
[INFO] Git: 5e551dd92b
[INFO] =================================================
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Overrides]: Redirecting save file to "/storage/savefiles/.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 45 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Mode 0: (3840x2160) 3840 x 2160, 30 Hz
[INFO] [DRM]: Mode 1: (4096x2160) 4096 x 2160, 60 Hz
[INFO] [DRM]: Mode 2: (4096x2160) 4096 x 2160, 60 Hz
[INFO] [DRM]: Mode 3: (4096x2160) 4096 x 2160, 50 Hz
[INFO] [DRM]: Mode 4: (4096x2160) 4096 x 2160, 30 Hz
[INFO] [DRM]: Mode 5: (4096x2160) 4096 x 2160, 30 Hz
[INFO] [DRM]: Mode 6: (4096x2160) 4096 x 2160, 25 Hz
[INFO] [DRM]: Mode 7: (4096x2160) 4096 x 2160, 24 Hz
[INFO] [DRM]: Mode 8: (4096x2160) 4096 x 2160, 24 Hz
[INFO] [DRM]: Mode 9: (3840x2160) 3840 x 2160, 60 Hz
[INFO] [DRM]: Mode 10: (3840x2160) 3840 x 2160, 60 Hz
[INFO] [DRM]: Mode 11: (3840x2160) 3840 x 2160, 50 Hz
[INFO] [DRM]: Mode 12: (3840x2160) 3840 x 2160, 30 Hz
[INFO] [DRM]: Mode 13: (3840x2160) 3840 x 2160, 25 Hz
[INFO] [DRM]: Mode 14: (3840x2160) 3840 x 2160, 24 Hz
[INFO] [DRM]: Mode 15: (3840x2160) 3840 x 2160, 24 Hz
[INFO] [DRM]: Mode 16: (1920x1080) 1920 x 1080, 120 Hz
[INFO] [DRM]: Mode 17: (1920x1080) 1920 x 1080, 120 Hz
[INFO] [DRM]: Mode 18: (1920x1080) 1920 x 1080, 100 Hz
[INFO] [DRM]: Mode 19: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 20: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 21: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 22: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 23: (1920x1080) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 24: (1920x1080i) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 25: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 26: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 27: (1920x1080) 1920 x 1080, 25 Hz
[INFO] [DRM]: Mode 28: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 29: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 30: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 31: (1360x768) 1360 x 768, 60 Hz
[INFO] [DRM]: Mode 32: (1152x864) 1152 x 864, 60 Hz
[INFO] [DRM]: Mode 33: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 34: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 35: (1280x720) 1280 x 720, 50 Hz
[INFO] [DRM]: Mode 36: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 37: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 38: (720x576) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 39: (720x576i) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 40: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 41: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 42: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 43: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 44: (720x400) 720 x 400, 70 Hz
MESA-LOADER: failed to open iris (search paths /usr/lib/dri)
failed to load driver: iris
MESA-LOADER: failed to open kms_swrast (search paths /usr/lib/dri)
failed to load driver: kms_swrast
MESA-LOADER: failed to open swrast (search paths /usr/lib/dri)
failed to load swrast driver
[WARN] [KMS]: Couldn't create GBM device.
[WARN] [DRM]: Couldn't get device resources.
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault

> lspci -vvv -d 8086:3184
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 605 (rev 03) (prog-if 00 [VGA controller])
	DeviceName: Onboard - Video
	Subsystem: Intel Corporation UHD Graphics 605
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 122
	Region 0: Memory at a0000000 (64-bit, non-prefetchable) [size=16M]
	Region 2: Memory at 90000000 (64-bit, prefetchable) [size=256M]
	Region 4: I/O ports at f000 [size=64]
	[virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express (v2) Root Complex Integrated Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 128 bytes, MaxReadReq 128 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis-, LTR-, OBFF Not Supported
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
			 AtomicOpsCtl: ReqEn-
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable- 64bit-
		Address: fee01004  Data: 4021
	Capabilities: [d0] Power Management version 2
		Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [100 v1] Process Address Space ID (PASID)
		PASIDCap: Exec- Priv-, Max PASID Width: 14
		PASIDCtl: Enable- Exec- Priv-
	Capabilities: [200 v1] Address Translation Service (ATS)
		ATSCap:	Invalidate Queue Depth: 00
		ATSCtl:	Enable-, Smallest Translation Unit: 00
	Capabilities: [300 v1] Page Request Interface (PRI)
		PRICtl: Enable- Reset-
		PRISta: RF- UPRGI- Stopped+
		Page Request Capacity: 00008000, Page Request Allocation: 00000000
	Kernel driver in use: i915

MrJustreborn, 2020-09-12:
I checked several nightly build (#1027 has the same issue)

20200910004523 <-> 20200907004317
The build from 7.sept has way more DRI libs installed

> i915_dri.so
i965_dri.so
iHD_drv_video.so
iris_dri.so
kms_swrast_dri.so
nouveau_dri.so
nouveau_drv_video.so
nouveau_vieux_dri.so
nvidia_drv_video.so
r200_dri.so
r300_dri.so
r600_dri.so
r600_drv_video.so
radeon_dri.so
radeonsi_dri.so
radeonsi_drv_video.so
s3g_drv_video.so
swrast_dri.so
vdpau_drv_video.so
virtio_gpu_dri.so
vmwgfx_dri.so
zink_dri.so

vs
> i915_dri.so
i965_dri.so
iris_dri.so
nouveau_dri.so
nouveau_drv_video.so
r300_dri.so
r600_dri.so
r600_drv_video.so
radeonsi_dri.so
radeonsi_drv_video.so

Ntemis, 2020-09-14:
https://bugs.launchpad.net/ubuntu/+source/mesa/+bug/1877879

Ntemis, 2020-09-19:
Please @MrJustreborn try this build and let me know
https://nightly.builds.lakka.tv/2020-09-19/Generic.x86_64/

MrJustreborn, 2020-09-21:
I've tested the live-version via USB-Stick and this build worked.

Ntemis, 2020-09-21:
Can we close this issue then?

closed 2020-09-21 by MrJustreborn