libretro/Lakka-LibreELEC #1031: The Granstream Saga crashes PCSX ReARMed

Cibomatto2002, 2020-09-14:
The game The Granstream Saga crashes PCSX ReARMed on Lakka Nightly 2020-09-13 Pie 4 2 gig model all you get is a black screen and it reboots back to RetroArch but Retroarch for windows 10 it loads and play fine. So far this is the only game I have trouble with but I only got 4 games tho.

Here is the log
[retroarch.log](https://github.com/libretro/Lakka-LibreELEC/files/5221448/retroarch.log)

Cibomatto2002, 2020-09-15:
Nevermind I found more and I'm giving up maybe one day there will be a better ps1 emulator for the pie 4

Ntemis, 2020-09-15:
I will try to find the game and try it

ToKe79, 2021-06-10:
please retest with [Lakka 3.x release](https://le.builds.lakka.tv) and open new issue if the problem persists.

closed 2021-06-10 by ToKe79

Cibomatto2002, 2021-06-10:
The game works on duckstation thanks for your time.