libretro/Lakka-LibreELEC #1035: Rock64 (RK3328) N64 black screen / crash upon game start.

WizardKnight, 2020-09-19:
I have a Rock64 (RK3328) that I tested with Lakka-RK3328.arm-3.0-devel-20200919030219-73ddb29-rock64.img.gz.
The SNES games seemed to work quite well. However the N64 games would show a black screen without any sound as soon as I tried to start them using either default emulator. The system stopped responding to the keyboard or joystick. I tried several games, and they all had the same result. I was forced to restart the system or the GUI.

I used systemctl stop retroarch.service ; LIBGL_DEBUG=verbose retroarch --menu --verbose >> log.txt 2>&1 from the terminal to get logs.

This is the output after that command:
`[INFO] RetroArch 1.9.0 (Git 5e551dd92b)
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4
[INFO] Built: Sep 19 2020
[INFO] Version: 1.9.0
[INFO] Git: 5e551dd92b
[INFO] =================================================
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Overrides]: Redirecting save file to "/storage/savefiles/.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 1 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 19 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Mode 0: (1920x1200) 1920 x 1200, 60 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 2: (1600x1200) 1600 x 1200, 60 Hz
[INFO] [DRM]: Mode 3: (1680x1050) 1680 x 1050, 60 Hz
[INFO] [DRM]: Mode 4: (1280x1024) 1280 x 1024, 75 Hz
[INFO] [DRM]: Mode 5: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 6: (1440x900) 1440 x 900, 60 Hz
[INFO] [DRM]: Mode 7: (1280x960) 1280 x 960, 60 Hz
[INFO] [DRM]: Mode 8: (1152x864) 1152 x 864, 75 Hz
[INFO] [DRM]: Mode 9: (1024x768) 1024 x 768, 75 Hz
[INFO] [DRM]: Mode 10: (1024x768) 1024 x 768, 70 Hz
[INFO] [DRM]: Mode 11: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 12: (800x600) 800 x 600, 75 Hz
[INFO] [DRM]: Mode 13: (800x600) 800 x 600, 72 Hz
[INFO] [DRM]: Mode 14: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 15: (800x600) 800 x 600, 56 Hz
[INFO] [DRM]: Mode 16: (640x480) 640 x 480, 75 Hz
[INFO] [DRM]: Mode 17: (640x480) 640 x 480, 73 Hz
[INFO] [DRM]: Mode 18: (640x480) 640 x 480, 60 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x40000001.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GL]: Vendor: ARM, Renderer: Mali-450 MP.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] Setting up menu pipeline shaders for XMB ... 
[INFO] [GLSL]: Compiling ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[... 227 lines cut ...]
[INFO] Disconnecting device from port 12.
[INFO] Disconnecting device from port 13.
[INFO] Disconnecting device from port 14.
[INFO] Disconnecting device from port 15.
[INFO] Disconnecting device from port 16.
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [Playlist]: Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] [GL]: VSync => on
[INFO] [Playlist]: Written to playlist file: /storage/.config/retroarch/content_history.lpl
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Game controller 0 (Standard controller) has a Memory pak plugged in
[libretro INFO] mupen64plus: Game controller 1 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Game controller 2 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Game controller 3 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Using CIC type X102
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
Segmentation fault
`
I am not sure what else I can do to help troubleshoot this.

Ntemis, 2020-09-19:
Thank you for the report it will be investigated

Ntemis, 2021-07-23:
Can you test v3.2 stable please

WizardKnight, 2021-07-23:
No I can not. I repurposed my rock 64 a long, long time ago. Sorry
-Sent from Android.

closed 2021-07-23 by Ntemis