libretro/Lakka-LibreELEC #1036: melonds unknown commit

brunnels, 2020-09-19:
Building current master fails at melonds because specified commit version doesn't exist.

```
❯ rm -rf sources/melonds
❯ ../build_opi_one_plus -e DISTRO=Lakka libreelec scripts/build melonds:target
GET      melonds (git)
    DELETE      (/build/sources/melonds/melonds-*)
    GIT CLONE      melonds
Cloning into '/build/sources/melonds/melonds-bb56907'...
remote: Enumerating objects: 12069, done.
remote: Total 12069 (delta 0), reused 0 (delta 0), pack-reused 12069
Receiving objects: 100% (12069/12069), 6.21 MiB | 2.95 MiB/s, done.
Resolving deltas: 100% (8544/8544), done.
There is no commit 'bb56907' on branch 'master' of package 'melonds'! Aborting!
```

Ntemis, 2020-09-19:
Yes the core has been updated and rebased on upstream branch.
Thank you for the report.
What system you are trying to build for?

brunnels, 2020-09-19:
Allwinner H6.  orange pi one plus.

Ntemis, 2020-09-19:
the new core is only for pc system and possibly aarch64

closed 2020-09-21 by Ntemis