libretro/Lakka-LibreELEC #1043: librga sources missing for OdroidGoAdvance

bhartshorn, 2020-09-24:
Hi. I am attempting to build Lakka for the OGA. It seems that the repository relied upon for [this librga package](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/graphics/librga/package.mk) has mysteriously disappeared: https://github.com/rockchip-linux/linux-rga

I looked around to see if it was moved or replaced, but I'm in over my head here. I did find rga sources in the rockchip-linux kernel tree, but I have no idea if it's the same code: https://github.com/rockchip-linux/kernel/tree/release-4.4/drivers/media/platform/rockchip/rga

Ntemis, 2020-09-24:
We wait to return

shantigilbert, 2020-09-24:
Meanwhile you can use this: https://github.com/Caesar-github/linux-rga

bhartshorn, 2020-09-24:
@shantigilbert thanks!
@Ntemis if it's known that it's coming back some time, feel free to close the issue. Thanks for response.

gouchi, 2020-10-23:
@bhartshorn [repository](https://github.com/rockchip-linux/linux-rga) is back.

You can also come to [discord](https://discordapp.com/invite/27Xxm2h).

closed 2020-10-23 by gouchi