libretro/Lakka-LibreELEC #1045: make errors and Python Traceback errors, cannot compile (ArchLinux)

cinnamonmatexfce, 2020-09-28:
ArchLinux

```
$ DISTRO=Lakka PROJECT=Generic ARCH=x86_64 VULKAN=yes VULKAN_SUPPORT=yes GRAPHIC_DRIVERS=nvidia make -j 16 image
```

```
make[4]: Entering directory '/home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/parallel-20190722/.x86_64-pc-linux-gnu/src'
rm /home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/toolchain/bin/sem || true
rm: cannot remove '/home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/toolchain/bin/sem': No such file or directory
ln -s parallel /home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/toolchain/bin/sem
make[4]: Leaving directory '/home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/parallel-20190722/.x86_64-pc-linux-gnu/src'
make[3]: Leaving directory '/home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/parallel-20190722/.x86_64-pc-linux-gnu/src'
make[2]: Leaving directory '/home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/parallel-20190722/.x86_64-pc-linux-gnu/src'
make[2]: Entering directory '/home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/parallel-20190722/.x86_64-pc-linux-gnu'
make[3]: Entering directory '/home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/parallel-20190722/.x86_64-pc-linux-gnu'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/parallel-20190722/.x86_64-pc-linux-gnu'
make[2]: Leaving directory '/home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/parallel-20190722/.x86_64-pc-linux-gnu'
make[1]: Leaving directory '/home/ctento/srcbuilds/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/parallel-20190722/.x86_64-pc-linux-gnu'
Traceback (most recent call last):
  File "scripts/genbuildplan.py", line 364, in <module>
    REQUIRED_PKGS = processPackages(args, ALL_PACKAGES, args.build)
  File "scripts/genbuildplan.py", line 290, in processPackages
    raise Exception(msg)
Exception: Invalid package reference: dependency yes in package beetle-psx::PKG_DEPENDS_TARGET is not valid
Parallel build failure - see log for details. Time of failure: Mon Sep 28 13:40:40 CEST 2020
make: *** [Makefile:12: image] Errore 1
```

Ntemis, 2020-09-28:
we dont support what you are trying to do here.Try the older branch(no support there either btw)

closed 2020-09-28 by Ntemis

cinnamonmatexfce, 2020-09-28:
Sorry... Then, is there a way I could get a working LAKKA with included:

- nVIDIA drivers
- Vulkan
- linux-firmware for kernel >5.1 (I need in order to get working WiFi+Bluetooth Intel® Wi-Fi 6 AX200 160MHz, see: https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi and https://www.intel.com/content/www/us/en/support/articles/000005511/network-and-i-o/wireless.html)

<br/>

BTW, LAKKA fails also in with defaults!
```
make[2]: *** [makefile.list:652: ErrorMsg.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/p7zip-16.02/CPP/7zip/Bundles/Alone'
make[1]: *** [makefile:17: 7za] Error 2
make[1]: Leaving directory '/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/p7zip-16.02'
FAILURE: scripts/build p7zip:host has failed!
[014/351] [FAIL] build   p7zip:host

The following logs for this failure are available:
  stdout: /Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/.threads/logs/28/stdout
  stderr: /Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/.threads/logs/28/stderr

parallel: This job failed:
package_worker 13 28 351 'build p7zip:host'
Parallel build failure - see log for details. Time of failure: Mon Sep 28 19:49:40 CEST 2020
make: *** [Makefile:12: image] Error 1
```

SethBarberee, 2020-09-28:
Assuming you are building on your host with `gcc-10`, try building with `gcc-9`.

cinnamonmatexfce, 2020-10-02:
> Assuming you are building on your host with `gcc-10`, try building with `gcc-9`.

I also removed `-j`, but still failing, same exact errors.
```
$ /usr/bin/gcc -v

Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc9/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-pkgversion='Arch Linux 9.3.0-3' --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,fortran,lto --enable-shared --enable-threads=posix --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --enable-cet=auto --program-suffix=-9 --enable-version-specific-runtime-libs --disable-multilib
Thread model: posix
gcc version 9.3.0 (Arch Linux 9.3.0-3)
```

```
/etc/makepkg.conf

[...]

#########################################################################
# ARCHITECTURE, COMPILE FLAGS
#########################################################################
#
CARCH="x86_64"
CHOST="x86_64-pc-linux-gnu"

#-- Compiler and Linker Flags
CPPFLAGS="-D_FORTIFY_SOURCE=2"
CFLAGS="-march=x86-64 -O2 -pipe -fstack-protector-strong -fno-plt"
CXXFLAGS="${CFLAGS}"
LDFLAGS="-Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now"
RUSTFLAGS="-C opt-level=2 -C target-cpu=native"
#-- Make Flags: change this for DistCC/SMP systems
#MAKEFLAGS="-j16"
#-- Debugging flags
DEBUG_CFLAGS="-g -fvar-tracking-assignments"
DEBUG_CXXFLAGS="-g -fvar-tracking-assignments"
#DEBUG_RUSTFLAGS="-C debuginfo=2"
```