libretro/Lakka-LibreELEC #1046: Thumbnails turn to black box

prince-rakeem, 2020-09-28:
- Which version of Lakka are you using? Both RPIi2.arm-2.3.2 and Lakka-RPi2.arm-3.0-devel-20200926034654-85aa8f4 have the same symptoms.  Logs are attached from Lakka-RPi2.arm-3.0-devel-20200926034654-85aa8f4 as I upgraded from 2.3.2 to the newest nightly to verify the issue is still reproducible.
- What system hardware are you using? Pi 3b+
- What did you do? navigated Lakka GUI for a few minutes.  
- What did you expect to happen? box art (thumbnails) for games to be visible.
- What happened instead? after a few minutes instead of showing box art there is a black box displayed.

This appears to be a similar issue a user was experiencing running RPIi2.arm-2.1.1 that was logged under bug "Images turn black" #[541](https://github.com/libretro/Lakka-LibreELEC/issues/541) which has been closed.

Log files and screenshot attached.
[image]
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/5293120/log.txt)
[log2.txt](https://github.com/libretro/Lakka-LibreELEC/files/5293122/log2.txt)

SethBarberee, 2020-09-28:
Do you have a way to reproduce? Is this for a certain game or all games on a certain platform?

prince-rakeem, 2020-09-29:
> Do you have a way to reproduce? Is this for a certain game or all games on a certain platform?

I did a bit more testing and it doesn't matter the platform or game.  All thumbnails turn blank.  I can reproduce by holding the up or down arrow to quickly scroll through the game list.  I also scrolled a bit slower to see if it was a specific game causing the issue, but that is not the case.  One thing I noticed and I'm not sure if it is related to the platform or the fact that I only have 2 games listed is for the DOOM platform.  They won't go black, but will disappear completely.  Scrolling slowly seems to bring them back in the DOOM platform, but not in the other platforms. The other platforms I have 10 or more games listed and all of them will cause the thumbnails to turn black after scrolling through the games for a while (after about 20-30 seconds).  Let me know if there is some other testing or logs that would be helpful.

Ntemis, 2020-09-29:
Seeing this be present on all the different retroarch versions that those images carry from 2.1.1 to 2.3.2 to latest nightly we will need some logs to be able to troubleshoot this better.

prince-rakeem, 2020-09-29:
> Seeing this be present on all the different retroarch versions that those images carry from 2.1.1 to 2.3.2 to latest nightly we will need some logs to be able to troubleshoot this better.

I attached 2 sets of logs to the initial issue.  If there is something more you need, please let me know which logs you are looking for and I can provide.

gouchi, 2020-09-30:
It must be related to this [issue](https://github.com/libretro/RetroArch/issues/6747) but we still need more investigation.

jdalmanza, 2021-02-24:
noticed the same issue on my rpi3 while curating my playlists. I found that if the thumbnail image is too large (larger than 1920x1080) it does not render the image and you get a "black" thumbnail. Same is true for graphical overlays. This is present on the most recent nightly builds last tested on devel-20210213015453-8076be1. Once I scaled the images down to 1080p or smaller, they displayed properly.

closed 2021-07-23 by Ntemis