libretro/Lakka-LibreELEC #1054: rpi4-aarch64: ppsspp fails build

basilean, 2020-10-08:
Hello guys,
I needed to add this line back in "projects/RPi/options" to complete image:
```bash
        LIBRETRO_CORES="${LIBRETRO_CORES// ppsspp /}"
```
As far I could see issue:
```bash
Lakka-LibreELEC/build.Lakka-RPi4.aarch64-3.0-devel/toolchain/lib/gcc/aarch64-libreelec-linux-gnueabi/9.3.0/../../../../aarch64-libreelec-linux-gnueabi/bin/ld.gold: error: cannot find -lpthreads
CMakeFiles/cmTC_5e9b3.dir/CheckFunctionExists.c.o:CheckFunctionExists.c:function main: error: undefined reference to 'pthread_create'
collect2: error: ld returned 1 exit status
```
It would looks like toolchain misses pthreads library.

I'm not sure if this is a problem of mine or a regression of:
https://github.com/libretro/Lakka-LibreELEC/pull/1050

Thank you.

dmrlawson, 2020-10-09:
It's been working for me, and it works on the build server. Have you tried a clean build (`make clean` before running again)?

basilean, 2020-10-09:
Yap.. I had this problem when I first cloned this repo, then, I made my private branch and cloned whole deal back to be sure.

I'm gonna give it a try later, if it works for everyone else, it should be a problem of mine. Thanks!

basilean, 2020-10-09:
It could be... or maybe just me... like this is a fresh effort, I preferred to report it.

I'm driving a T440p powered by "Debian GNU/Linux bullseye/sid"

basilean, 2020-10-14:
I just got a fresh clean aarch64 build, I suspect cloned git was corrupt, I should have started caching.
Sorry for false alarm.

closed 2020-10-14 by basilean