libretro/Lakka-LibreELEC #1056: Please add Vulkan video driver for rpi4-aarch64

SpaceAgeHero, 2020-10-08:
Now that the first 64 bit nightly builds are available for Raspberry Pi 4, I'd love to use DuckStation.
However for that Vulkan is missing. Can you please add it?

basilean, 2020-10-09:
I gave a try using this WIP driver:
https://gitlab.freedesktop.org/apinheiro/mesa/-/tree/wip/igalia/v3dv

In order to do so, I needed to update Meson:

- packages/python/devel/meson/package.mk

```bash
PKG_VERSION="0.55.3"
PKG_SHA256="6bed2a25a128bbabe97cf40f63165ebe800e4fcb46db8ab7ef5c2b5789f092a5"
```
- Remember to update toolchain:

```bash
DISTRO=Lakka PROJECT=RPi DEVICE=RPi4 ARCH=aarch64 scripts/build meson:host
DISTRO=Lakka PROJECT=RPi DEVICE=RPi4 ARCH=aarch64 scripts/install meson:host
```
Later, I needed to update mesa for this arch:

- packages/graphics/mesa/package.mk

```bash
...
if [ "$TARGET_ARCH" = "aarch64" ]; then
  PKG_VERSION="5d537f57"
  PKG_SHA256="7c0ec31b5cb174414a027cbe8b827e2e3720f10bdcf2d546aeea142608be0ca3"
  PKG_URL="https://gitlab.freedesktop.org/apinheiro/mesa/-/archive/wip/igalia/v3dv/mesa-wip-igalia-v3dv.tar.gz"
fi
...
...
if [ "$TARGET_ARCH" = "aarch64" ]; then
  PKG_MESON_OPTS_TARGET="${PKG_MESON_OPTS_TARGET//-Dvulkan-drivers=auto/-Dvulkan-drivers=broadcom}"
fi
```
It compiles and I can see this library on the image:

```bash
/usr/lib/libvulkan_broadcom.so
```
Now, I'm getting stuck trying to enable it in retroarch:

- packages/libretro/retroarch/package.mk

```bash
  if [ "$DEVICE" == "RPi4" ]; then
    RETROARCH_GL="--enable-vulkan --enable-opengles --enable-opengles3 --enable-opengl_core --enable-kms --disable-x11 --disable-videocore --enable-plain_drm"
```

```bash
Forced to build with library -lvulkan, but cannot locate. Exiting ...
```

It looks like I'm missing something in the toolchain but I can't get it.
Any help would be much appreciate it.
Thanks.

dmrlawson, 2020-10-10:
@basilean I'm working on it here: https://github.com/dmrlawson/Lakka-LibreELEC/tree/rpi4-vulkan

I've got those changes, you're missing `vulkan-loader` and `vulkan-headers`. I can build it but retroarch segfaults on startup at the moment.

basilean, 2020-10-11:
Hey man... nice one... you already got it!
yeah... I had to spend few hours on khronos git following first unix piece of advice... RTFM! =P

One thing that gave me a clue but got me confused is:
retroarch/package.mk:51
```bash
 PKG_DEPENDS_TARGET+=" $VULKAN vulkan-loader"
```
I never found package "vulkan-loader" so I added it, it has an update_deps.py script that take cares of the headers to compile.

Then I created a "broadcom" virtual package just with deps on brcm driver, mesa and vulkan-loader.

Finally just adding VULKAN="broadcom" on rpi options.

It was pretty disappointed to see it crash when I switched video drivers but I keep hope that one day is going to work. =D

basilean, 2020-10-11:
@dmrlawson - I took a look to your code, one thing I made different is that I took out X support instead of add libraries.
I would understand that is not needed
APP -> MESA -> DRM KERNEL
https://upload.wikimedia.org/wikipedia/commons/2/2d/The_Linux_Graphics_Stack_and_glamor.svg
```bash
PKG_CMAKE_OPTS_TARGET=" -DBUILD_WSI_XLIB_SUPPORT=OFF "
```

Following this game... I wanted to go a little bit deeper... so I added "vulkan-tools"... in order to compile I stripped all X related support...
```bash
PKG_CMAKE_OPTS_TARGET=" -DBUILD_WSI_XLIB_SUPPORT=OFF -DBUILD_WSI_WAYLAND_SUPPORT=OFF -DBUILD_CUBE=OFF -DBUILD_WSI_XCB_SUPPORT=OFF "
```
Just wanted to have a simple talk with the stack... of course I got a sigsegv. =P

A first look with strace shows that it uses all components as expected
```bash
...
openat(AT_FDCWD, "/usr/lib/libvulkan.so", O_RDONLY|O_CLOEXEC) = 3
...
openat(AT_FDCWD, "/usr/share/vulkan/icd.d/broadcom_icd.aarch64.json", O_RDONLY) = 3
...
openat(AT_FDCWD, "/usr/lib/libvulkan_broadcom.so", O_RDONLY|O_CLOEXEC) = 3
...
openat(AT_FDCWD, "/usr/lib/libdrm.so.2", O_RDONLY|O_CLOEXEC) = 3
...
openat(AT_FDCWD, "/dev/dri/renderD128", O_RDWR|O_CLOEXEC) = 3
```

Going a little bit hardcore... I set to get a coredump
```bash
ulimit -c unlimited
echo "/storage/%e.%s.core" > /proc/sys/kernel/core_pattern
```

Tacking a look to backtarce with gdb
```
...
#4  0x0000007fa3054464 in vkEnumeratePhysicalDevices () from /usr/lib/libvulkan.so
...
```
Putting both together... I would guess that mesa broadcom is not talking with hardware as expected.

basilean, 2020-10-12:
One more to finish the day...

After few changes on mesa:
```bash
                       -Dbuild-tests=true \
...
    PKG_MESON_OPTS_TARGET+=" -Dplatforms=drm -Ddri3=true -Dglx=disabled"
```

I got a clear error:
```bash
ERROR at ../vulkaninfo/vulkaninfo.h:248:vkEnumerateInstanceExtensionProperties failed with ERROR_INITIALIZATION_FAILED
```

I'm not sure if this helps anyone else efforts but at least I'm getting fun. =D

basilean, 2020-10-12:
I couldn't resist a last debug...

```bash
...
       773:	calling init: /usr/lib/libvulkan_broadcom.so
       773:	
       773:	/usr/lib/libvulkan_broadcom.so: error: symbol lookup error: undefined symbol: vk_icdNegotiateLoaderICDInterfaceVersion (fatal)
       773:	/usr/lib/libvulkan_broadcom.so: error: symbol lookup error: undefined symbol: vk_icdNegotiateLoaderICDInterfaceVersion (fatal)
       773:	/usr/lib/libvulkan_broadcom.so: error: symbol lookup error: undefined symbol: vk_icdNegotiateLoaderICDInterfaceVersion (fatal)
       773:	/usr/lib/libvulkan_broadcom.so: error: symbol lookup error: undefined symbol: vk_icdNegotiateLoaderICDInterfaceVersion (fatal)
...
```

dmrlawson, 2020-10-12:
@basilean have you committed this to github somewhere? Feel free to make a PR to my branch or link to yours

basilean, 2020-10-12:
hey @dmrlawson... no... at this point is such a shotgunning that I'm not even sure they are going to work straightforward... I'm going to start clean... checking that every little piece work as expected... to be true.. it was more than what I could chew in a couple of days....

Founding so much fuzzy information related to aarch64+vulkal... What I looked for is to share my effort with everyone... less skilled would found it useful... more skilled could teach me something new. =)

1 - I tried to fully learn how LibreELEC framework works... I had little experience on it from a time ago...  so I read readme bellow and took some notes about workflow. (I would like to share, maybe forum?)
https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/readme.md

2 - Understanding how the linux kernel graphic stack works with mesa... or at least give some meaning on what I was trying to accomplish...
https://en.wikipedia.org/wiki/Mesa_(computer_graphics)

3 - Learning about how to build vulkan... from here I got flags to disable x related and added debug...
https://github.com/KhronosGroup/Vulkan-Headers/blob/master/BUILD.md
https://github.com/KhronosGroup/Vulkan-Loader/blob/master/BUILD.md
https://github.com/KhronosGroup/Vulkan-Tools/blob/master/BUILD.md

4 - Checking broadcom implementation updates... it is only trusted source about it... it looks like they are not interested on 64 or newer api...
https://gitlab.freedesktop.org/apinheiro/mesa/-/issues
https://gitlab.freedesktop.org/apinheiro/mesa/-/commits/master

5 - Applying some debug tricks to get a clue why were not working...
- strace to see system calls and check it were opening files needed... the one in packages were not building so I took borrowed the binary from a debian aarch64 package.
```bash
strace ./vulkaninfo
```

- gdb to see under the hood when it was crashing...
```bash
gdb ./vulkaninfo
...
run
...
bt
...
frame X
```

- Useful environments variables to get verbose output and force locations to not update whole image.
```bash
LD_DEBUG=libs ./vulkaninfo
```
```bash
VK_LOADER_DEBUG=debug ./vulkaninfo
```
```bash
VK_ICD_FILENAMES=broadcom.json ./vulkaninfo
LD_LIBRARY_PATH=/storage ./vulkaninfo
```

So far my conclusion is that broadcom implementation uses an old way of initialize so it could be a shot using older headers/loader sources (I would expect backwards compatibility so not sure if would work) or start contributing on that repo for the missing pieces.
"vkEnumerateInstanceExtensionProperties" is present on amd and intel drivers but not in broadcom.

I'm disappointed that it didn't work but at least I learned couple of thing and left all pieces together on this post. =D

dmrlawson, 2020-10-13:
Looks like it just got added: https://gitlab.freedesktop.org/apinheiro/mesa/-/commit/f5ce89808a52c2fa4b20a664ee465566e58b5c14

I've somewhat given up for now, though it might be worth seeing if that makes a difference

basilean, 2020-10-13:
hey @dmrlawson... I just gave it a try... while this time there is no fatal... still having issues at init...

```bash
nx1701:~ # VK_LOADER_DEBUG=debug vulkaninfo
...
ERROR at ../vulkaninfo/vulkaninfo.h:248:vkEnumerateInstanceExtensionProperties failed with ERROR_INITIALIZATION_FAILED
...
```

One interesting thing I found:
https://github.com/Yours3lf/rpi-vk-driver

This nvidia guy created a rpi1,2,3 vulkan driver just for fun... he uses a different approach... if that works... it could be a live extender for old hardware.

basilean, 2020-10-14:
@dmrlawson commits got pilled on this pr... I got a clean build... just commenting out:

RPi/options:54
```bash
        # VULKAN="mesa"
```
It should leave the image at the point I reported above.
I will give it a try once in a while hopping one day found it working.

https://github.com/libretro/Lakka-LibreELEC/pull/1055

dmrlawson, 2020-10-15:
Based on [this comment](https://blogs.igalia.com/itoral/2020/07/23/v3dv_vulkan_driver_update/#comment-26161) I think it's not expected to work at the moment, which is disappointing. The driver is now merged into mesa though so perhaps it will get done sooner rather than later.

The only option to try right now is to make a Lakka build using x11. I've tried quite a bit but struggle to get xorg-server to compile.

Otherwise we just have to wait :-\

RaydenX93, 2020-10-22:
https://www.raspberrypi.org/blog/vulkan-update-merged-to-mesa/

dmrlawson, 2020-11-12:
Since https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/7515 was merged we can now try this again.

I've added it to the code in a bit of a hacky way, but we can test it until there's a new release of mesa that includes it.

I had terrible fps until I disabled v-sync and enabled threaded video in retroarch, but with those settings I get ~60fps in Duckstation with the vulkan renderer.

Code:
https://github.com/dmrlawson/Lakka-LibreELEC/tree/rpi4-vulkan

Images:
https://drive.google.com/drive/folders/1THDW4Py1KepUv4X572gr7ON62hlcl83w

basilean, 2021-01-20:
Hey @dmrlawson , guys,
 I was AFK for a couple of months, its so nice to see this topic has moved on, feels like the dream of the best and cheaper console is getting close. =D

Yesterday, I compiled using latest git for mesa and vulkan-*, so far:
- flycast is working like a champ adding vulkan flag as you did.

- dolphin, the one I'm more interested on is not working for me. Does it works for you?
  I saw libretro branch is far behind original where they added a vulkan flag.
  I was thinking like a shoot to try using original repo and manually applying libretro patches.

- parallel is also not working, actually, using gl, I only got rice plugin to work more or less decently, parallel hangs.

- play not working with vulkan nor gl, looks like pretty girl you want to invite dance. =P

Nowadays I'm back to reality and warming up but I would like to understand right way to use git on this project and collaborate more often (please anyone to write a howto for dummies). IE: These packages are broken: psp1, libretro-mame2016, RTL8812AU

Happy new year everyone!

SpaceAgeHero, 2021-01-21:
I've recently tried Retroarch with Vulkan on Raspberry Pi 4 Manjaro ARM AARCH64 and was disappointed by the performance to be honest.
Maybe I did something wrong but there was no real gain in DuckStation contrary to GL.
Hopefully it will be better without the OS overhead on Lakka.

Ntemis, 2021-03-27:
Vulkan is now up and running.
We can safely close this.

closed 2021-03-27 by Ntemis