libretro/Lakka-LibreELEC #1060: Piboy DMG support to master

acidtech, 2020-10-13:
I am posting to see if any of the Lakka developers would like to add Piboy DMG support to Lakka. I can provide the necessary hardware and sources to our kernel module driver and OSD.  

I have created a custom image that works with our hardware but is now out of date(I made it last year for testing) and I will be too busy to do more work on it for at least three more months.  I would also much prefer having it added to the master build instead of providing custom images to our users.

If you are interested please contact me at support@experimentalpi.com

Thank you.

Ntemis, 2020-10-13:
We can add it for sure but are you willing to maintain it if it breaks?
Our hands are full as it is.

basilean, 2020-10-13:
@acidtech... I just google it... is soooo cute... congrats!
With not much effort... Lakka using LibreELEC building/branding/booting/updating system is the best distro around for it.

Where are specs about display and input? I would understand just enabling right boot overlays and kernel modules it should work out of the box.

acidtech, 2020-10-14:
If you add it I will maintain it.  I did it with the old LE9.2 last year
and it worked great.  But I havent touched it in so long I forgot
everything I did.  For me it would take days to get everything back up and
running with master and then debugged.  If you do that work I can keep the
kernel module and osd app up to date.

Nathan Scherdin

acidtech, 2020-10-14:
Display is DPI LCD mode 6(18bit color).  Everything is set though the
config.txt so that part is easy.  Audio is moved to GPIO18/19.  Kernel
driver communicates through GPIO26 and 27.  GPIO10 and 11 are free.

You can grab all the software we use with Retropie here:
https://www.experimentalpi.com/Installing-the-Safe-Shutdown-and-OSD-Software-on-the-PiBoy-DMG_b_21.html

This works fine for Retropie but not so much Lakka of course.  :)

The basic system is:
payload/usr/src/xpi_gamecon: kernel module source code and DKMS build for
interfacing with the Piboy controls.  For lakka this needs to be changed to
a standard make file build for a kernel module.

payload/home/pi/osd: handles on screen display as well as fan control via
cpu temp.

payload/lib.systemd/system/xpi_gamecon_shutdown.service: A service to shut
down the osd executable if a software shutdown happens.  Shutting down the
OSD will send a signal to the Piboy to power down after a delay(eg Pi stops
communicating, Piboy waits X seconds and powers everything down.

I dont know if you want all this here.  We can work out details of who
needs a unit via email, eg for shipping addresses and such.

Contact me at support@experimentalpi.com for that information.

Let me know if you need anything else.

Nathan Scherdin

basilean, 2020-10-14:
@acidtech I'm seeing that you offer a package to apply to a standard retropie image installed... right?
If so... why don't do same with Lakka?

You just need to make a tweak like this one:
https://forum.libreelec.tv/thread/14490-implementing-xpadneo-drivers-for-future-releases/?postID=121042#post121042

systemd configuration is placed under permanent storage so not big deal there. Same for rpi config.

If you wanna go wild... as Lakka uses best of LibreELEC... you can get best of both... branch... and add a distribution with your own brand.

acidtech, 2020-10-14:
I'm using Ubuntu(via VMware) and cross compiling for lakka.  I dont see how
I get the proper header files using the method you linked.  The way I did
it before was to clone the Lakka repo, add my packages into the packages
directory, add some lines to the appropriate script to build my packages
and then built everything.  Takes forever.

It was similar to what you linked but I build everything basically creating
my own release like you suggested.  Its been more than a year now so there
was more to it and it took a while to figure out since it was the first
time I was doing anything with LibreElec but that was the gist.

Nathan Scherdin

basilean, 2020-10-15:
Right... I made it using build system...  "scripts/build PKG"... then extracted module.ko... You need to do it each time kernel version changes.
You could package it in a zip with autostart script, config files, a oemsplash.png and a small install script. Users, once installed a release image would need just to connect by ssh hit a wget, unzip, ./install.sh and reboot.

acidtech, 2020-10-15:
Cool. I think I understood most of that. :)  I will probably be back with
questions after I attempt this over weekend.

Nathan Scherdin

ShockwaveTheFallen, 2020-11-27:
Any word on this yet? :)

Paulie420, 2021-03-18:
Or yet? I want to see the PiBoy DMG on other OSes-  Lakka would be cool. And then... raspberry pi os..Manjaro..twister.

acidtech, 2021-03-19:
Eventually.  OSD still has some work. Once that is done I will open source
it.  If a third party wants to work on any of these now they can. The
kernel module is already open source and it has all that is needed.  The
OSD just gets data from it to display and sends settings from the osd.cfg
back to it via the /sys/kernel/xpi_gamecon virtual files.  Also if someone
is interested I will supply the OSD source as is to anyone who wants to
work on any of these OS's.

The main catch you will have is fKMS vs KMS.  OSD uses dispmanx currently.
And I have not found any way to do an overlay for on screen icons using KMS
without using X.

Nathan Scherdin

duduke, 2021-06-10:
Any chance for a guide so that someone can create their own PiBoy DMG LAKKA build?

acidtech, 2021-06-11:
There is no easy way to do it.  The way it is normally done is to get the
Lakka build environment(eg source code), add our kernel driver and osd
software and then build a Lakka image.  We gave away several units so some
groups would do this but they have not, yet.  Seems to be taking longer
than expected.

Another option would be to extract the lakka compressed system image,
modify it(add our software) and recompress it.  Lakka uses a special
compressed OS.  I knew how to do that a couple years ago but I have not
looked into it for a while and cant remember the procedure anymore. I just
know it can be done.

Nathan Scherdin

ToKe79, 2021-06-14:
so basically we could use the rpi kernel tree? we currently use:
https://github.com/libretro/Lakka-LibreELEC/blob/fe6b66a3db142d38e9c4b1c93ff9342f71cf5dbd/packages/linux/package.mk#L30-L37
can you provide sources for the driver and osd software? also are any special changes required for the kernel config? or we can use the ones we already use for RPi devices? The end-user probably can use any of those SBCs as the internals, so this would mean we would need to build separate versions for each RPi, right?

duduke, 2021-06-14:
I've had a quick look (I have no relation to experimental-pi), and the source for the driver and accompanying software is located here: https://downloads.experimentalpi.com/PiboyDMGUpdate1.03.zip.
Within that zip, the kernel driver is located in **PiboyInstall/payload/usr/src/xpi_gamecon-1.0** and looks like it can be easily adapted to compile as a LAKKA additional package.
The OSD under software under **PiboyInstall/payload/home/pi/osd** looks like an ARM binary, but I can't see a source code for it.
Maybe some of the functions of the OSD software can be diverted to LAKKA - I know LAKKA can display battery status. 
Fancontrol seems a bit more trickier...

duduke, 2021-06-18:
@acidtech can I write values to /sys/kernel/xpi_gamecon to drive the volume/fan if so, can you give some examples? 

Thanks.

acidtech, 2021-06-18:
Yes. That is the user mode interface to the kernel module.  Values are just
0 to 100 for volume.  Fan is 0 to 255(equates to 0 to 25.5 percent duty).
Written as ascii text to the file.  You can edit them manually with sudo
nano to experiment.

I would grab the 1.04 beta though.  It has all the same files as the 1.03
update but has a lot of fixes in it.  It's been well tested.  We havent
made it the new release yet just because I've been busy working on the 1.05
release.

http://downloads.experimentalpi.com/ExperimentalPiUpdate_Beta.zip

The OSD has no source code available yet.  There is a ton of work I have to
do before I feel comfortable releasing source for that.  I expect you could
customize Lakka to do basically everything the OSD does.  In the long run
that may be the best option because eventually KMS will become the standard
and there is no easy way to handle a OSD via a third party app in KMS.

Nathan Scherdin

duduke, 2021-07-04:
I've been able to make some progress over the past couple of days:
1. I've got LAKKA booting on the PiBoy by merging config.txt settings from ExperimentalPi with distroconfig.txt that ships with LAKKA.
I guess this needs further cleanup, but here it is: [https://github.com/duduke/xpi_gamecon/blob/main/distroconfig.txt](url)
2. I've got the xpi_gamecon to compile via a rudimentary package.mk [https://github.com/duduke/xpi_gamecon/blob/main/package.mk](url) the package sources the code from the beta posted above which i'm hosting at my git for the being.
I've built and pushed the resulting xpi_gamecon.ko to /storage/.config and loaded it via insmod. Lakka then detects it but, button mapping is wrong.
3. I'll start working on somehow controlling the fan via a simple service.

As I have no knowledge of working with GIT, can someone turn this into a PR and review my files to start working on a LAKKA version for PiBoy DMG?

Thanks.

duduke, 2021-07-05:
Got a bit more progress:
I've created a simple shell script that can run in the background and monitors cpu_temp (every 5 secs) and pushes a value to /sys/kernel/xpi_gamecon/fan.
@acidtech Fan values are 0-255, how do I push a 100% value to the fan? is it 100 or 255?
Volume seems a bit more complicated as I would need to read the value from /sys/kernel/xpi_gamecon/volume, but I don't yet know where to write it to the OS.
Battery is on another level as on PiBoy DMG currently hosts battery information at /sys/kernel/xpi_gamecon/, but LAKKA doesn't read from there...
A long way to go still....

acidtech, 2021-07-05:
255 equals 100%.  0 thru 254 equals 0.0 to 25.4 percent duty.  With all
fans we have shipped with there is no advantage going over 25.4% duty but
the 100% flag(255) is there just in case.  Most fans are anywhere from 7
percent(70) to 15 percent(150).  Anything lower on most fans and they stop
or wont start.  Anything higher and the fan doesnt gain any
noticeable speed.

On Retropie we use the amixer command line to set the volume(eg for
alsamixer).  You will need to find out of Lakka is using alsa audio or not.
If it is you can do the same.

Please keep me informed on your progress.  When you get to the point you
are ready to play with the OSD let me know.  I believe lakka is still use
fKMS so the OSD should be able to be recompiled without significant
changes.  If they have moved to KMS, well, we will have to figure something
else out.  I use dispmanx for the OSD graphics and KMS deprecates that.

Nathan Scherdin

duduke, 2021-07-05:
@acidtech I'm not entirely sure OSD is a must for LAKKA. 
In summary:
1. I've created a simple script that monitors VOLUME and FAN/TEMP and can be used as a systemd service - [https://github.com/duduke/xpi_gamecon/blob/main/xpi_gamecon-tool](url)
2. xpi_gamecon driver successfully compiles with LAKKA - I've created a relevant package.mk file - [https://github.com/duduke/xpi_gamecon/blob/main/package.mk](url) to make it work, i'm hosting the latest xpi_gamecon beta at my git.
3. I've created a joypad autoconfig file for PiBoy DMG - [https://github.com/duduke/xpi_gamecon/blob/main/PiBoy%20DMG%20Controller.cfg](url)
4. I've merged the settings from PiBoy retropie config.txt to LAKKA's distoconfig.txt - [https://github.com/duduke/xpi_gamecon/blob/main/distroconfig.txt](url)
5. @acidtech - When I switch the power button, the device seems to shutdown correctly. is that true? or is something else needed for proper shutdown?

I think that at this stage, it's most of the important pieces are in place. I'd be happy if someone from the LAKKA dev team can create a PR from those changes as I'm pretty much clueless where it comes to github. @Ntemis?

acidtech, 2021-07-05:
5. Yes but we have two systemd scripts. One for shutdown and one for
reset(shutdown tells piboy to shutdown quick instead of timing out on loss
of communications to the RPi and restart tells Piboy NOT to shutdown at
all, to handle restarts).  Look in the SDInstall/payload/lib/systemd/system
folder in.

http://downloads.experimentalpi.com/ExperimentalPiUtility.zip

The shutdown script isnt required though it makes shutdown cleaner.  The
restart script is required if you want to handle restarts properly.

Ignore the other scripts in that folder. Those are used for the automatic
installer.

How do you plan on dealing with battery notification without an OSD?

Nathan Scherdin

duduke, 2021-07-06:
AFAIK, Lakka already has instrumentation in place to display battery status (several devices already do: RG351, laptops etc). So either adapt Lakka to read battery status from the xpi_gamecon driver. Or, alternatively have xpi_gamecon driver publish battery status to a "standard" location, such as: /sys/class/power_supply/BAT0/
If battery is taken care of, I don't see an immediate need for the OSD software.

dmrlawson, 2021-07-06:
Hi :wave: 

I'm going to try to look at this later after work and start a WIP PR with the bits that you've already done. I think we use KMS so for the OSD I think the easiest thing is to hook into the built-in stuff (though I've not touched that before personally).

Which RPis do you want to support, and which architectures? We have separate targets for e.g. rpi4-arm, rpi4-aarch64, rpi3-aarch64 etc so depending on your requirements we might need to create multiple targets.

duduke, 2021-07-06:
This is probably more for @acidtech, but to my knowledge, PiBoy DMG supports pi0, pi3, pi4.
I've only tested on Pi4 with lakka compiled for aarch64.

dmrlawson, 2021-07-06:
@acidtech are you willing to be the maintainer for this?

acidtech, 2021-07-06:
The later is very easy.  Just change the path in the kernel module for the
battery virtual file.  Will need to adjust the formatting to match whatever
is required in BAT0.

Nathan Scherdin

acidtech, 2021-07-06:
Yes. Piboy supports all those but people using Pi0 with it has been
basically zero. I wouldnt waste any effort on that.  The Pi3 has sold well.
The Pi4 of course has been the majority but we still have sold units for
Pi3s quite well.

Nathan Scherdin

acidtech, 2021-07-06:
I have zero experience with github myself.  I am more than willing to work
with anyone that can handle the nuts and bolts of github/submitting updates
to Lakka.  I have already built a custom Lakka build to test Lakka on
Piboy.  It is now very far out of date but I can handle the coding side if
someone else can handle the github/lakka sibmitions.

Nathan Scherdin

acidtech, 2021-07-06:
Currently the kernel module works with 32bit only.  Some changes will be
needed to support aarch64.

Nathan Scherdin

dmrlawson, 2021-07-06:
Ok, if you're happy to do the coding then I suggest downloading the current master branch (https://github.com/libretro/Lakka-LibreELEC/archive/refs/heads/master.zip), making your changes and I can push it to github.

duduke, 2021-07-07:
@acidtech I was able to compile xpi_gamecon for aarch64 without any changes.
resulting driver shows this when running "file":
`./xpi_gamecon.ko: ELF 64-bit LSB relocatable, ARM aarch64, version 1 (SYSV), BuildID[sha1]=a6f7172a55563704a496dc12d4826db737307c95, not stripped`

acidtech, 2021-08-04:
> @acidtech <https://github.com/acidtech>
> I'm getting those errors in syslog:
> XPi Gamecon CRC Error: 1534 0001
> XPi Gamecon CRC Error: 0092 0001
> XPi Gamecon CRC Error: 0291 0001
> XPi Gamecon CRC Error: 0436 0001
> XPi Gamecon CRC Error: 0024 0001
> XPi Gamecon CRC Error: 0398 0001
> Everything seems to work OK, so I don't yet know the reason for those
> errors.

Those errors are normal with the current firmware.  They indicate the
number of good packets since the last bad packet and then the number of
failed packets. Communications is handled in a RPi timer interrupt
processing at 100hz.  Piboy is expecting communications regularly at
100hz.  Piboy has a timeout to detect potentially bad packets(clears the
packet buffer if x ms pass with no data from the RPi).  RPi can take longer
to process the timer interrupt than the Piboy timeout allows, causing a
"failed" packet.  This is normal behavior.  As long as you aren't getting
huge numbers of bad packets there is no problem.  The number of errors will
go up under heavy load(see results after playing a heavy game) but I've
never seen anything that prevents at least 60 updates a second.

Nathan Scherdin

duduke, 2021-08-08:
@acidtech Thanks for the info.
I've got some more progress: i've added the shutdown and reboot systemd scripts to LAKKA, and now shutting from within retroarch, does a clean poweroff.
Which value should I read from /sys/kernel/xpi_gamecon to detect that the power button was switched and force a system shutdown?
P.S. I'm still working on having the battery show up in retroarch.

duduke, 2021-08-09:
@acidtech, i'm hitting a brick wall with getting the battery to display via the standard power_supply class (https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-class-power)
I think the difference is because you are using kobject_create_and_add, and this requires class_create so it's not as simple as renaming the path.
Also, i've never did any C coding (only some scripting), so I might be missing something...

acidtech, 2021-08-10:
xpi_gamecon creates a /sysfs file you can read the shutdown status from.
/sys/kernel/xpi_gamecon/stat

Read the value into a byte(eg a value from 0 to 255).  The high bit(bit 7)
indicates VBus is plugged in(eg 0x80 mask), bit 6 indicates the power
switch state(eg 0x40 mask).  Any other bits can be ignored with the current
firmware.

This is the code I run based on the power switch bit:
if((state.status&0x40)==0){
   system("/home/pi/osd/multi_switch.sh --es-poweroff &");
   break;
}

multi_switch.sh is a trimmed down version of the multi_switch shutdown
script commonly used with retropie.

Nathan Scherdin

duduke, 2021-08-13:
@dmrlawson @acidtech 
Hi,

Over the past couple of days, I've been able to get most things working. All the relevant code is in this repo: https://github.com/duduke/piboy-dmg-lakka-test/tree/main
I've created a small guide: https://github.com/duduke/piboy-dmg-lakka-test/tree/main#readme

The way I understand it, the following are needed to create a LAKKA device for PiBoy DMG:
1. A `package.mk` for the PiBoy DMG kernel module (https://github.com/duduke/piboy-dmg-lakka-test/blob/main/package.mk). 
This needs to be reviewed, and probably changed when creating a device profile for LAKKA. As it is, i've been able to compile it under LAKKA with the device set to RPi4 AARCH64.
2. A control script to replace the capabilities of the OSD software (https://github.com/duduke/piboy-dmg-lakka-test/blob/main/piboy-dmg-control.sh)
The script manually loads the PiBoy DMG kernel module. This line: `insmod /storage/roms/piboy-dmg/xpi_gamecon.ko` should be removed when creating an official LAKKA device.
3. A systemd service that activates the script (https://github.com/duduke/piboy-dmg-lakka-test/blob/main/piboy-dmg-control.service). 
Again, this should be edited to reflect the PiBoy DMG Control script location when creating an official LAKKA device.
4. A `distroconfig.txt` file specific to PiBoyDMG (https://github.com/duduke/piboy-dmg-lakka-test/blob/main/distroconfig.txt)
5. A PiBoy DMG controller profile (https://github.com/duduke/piboy-dmg-lakka-test/blob/main/PiBoy%20DMG%20Controller.cfg) the file should be placed in `/storage/joypads/udev` and be called `PiBoy DMG Controller.cfg`

The only thing that's currently missing is battery percentage monitoring, which can be accomplished either:
1. PiBoy kernel module writing battery status to /sys/class/power_supply/BAT0
Or:
2. Retroarch read battery percentage from /sys/kernel/xpi_gamecon/percent.

Maybe this is enough for an alpha version, and then someone more gifted than me can improve on it?

dmrlawson, 2021-08-14:
@acidtech are you going to put this into the Lakka tree for me to then add to git?

I'm unwilling to do more than that without having the hardware myself to test

acidtech, 2021-08-14:
If you can get this put into the Lakka tree we are more than willing to
send you an assembled Piboy.  Let me know.

Nathan Scherdin

acidtech, 2021-08-14:
I looked into the linux power supply driver(eg for BAT0).  Looks like that
could be integrated into the xpi_gamecon kernel module but its not a simple
project.  I'll have to do some reading to see what the minimum
functionality required is.

Nathan Scherdin

dmrlawson, 2021-08-15:
> If you can get this put into the Lakka tree we are more than willing to send you an assembled Piboy. Let me know. Nathan Scherdin
> […](#)

I've just sent an email to the address in the first comment

duduke, 2021-08-15:
I've added a custom boot splash to be placed in LAKKA boot partition (oemsplash.png):
https://github.com/duduke/piboy-dmg-lakka-test/blob/main/oemsplash.png

acidtech, 2021-08-19:
Send to ***@***.***  That will come to me and I'll get it
forwarded to the right people. The expi emails are all forwarded to
autoresponders because everything is going through the CRM ticket system
now.

Nathan Scherdin

dmrlawson, 2021-08-20:
> […](#)

I think github filters email addresses. Are you able to join the discord? https://discord.gg/BNFR4hM

dmrlawson, 2021-09-07:
@duduke please will you test the latest nightly for PiBoy DMG and confirm it works for you? I've added all the work you did. The current nightly (2021-09-07) sets the default menu to ozone but I've got a PR to change it to rgui as I think it works better on this screen.

duduke, 2021-09-08:
> @duduke please will you test the latest nightly for PiBoy DMG and confirm it works for you? I've added all the work you did. The current nightly (2021-09-07) sets the default menu to ozone but I've got a PR to change it to rgui as I think it works better on this screen.

I'll test later today. 
I think we should keep Ozone. If the "Menu Scale Factor" is set to 1.50x Ozone looks and works great on the PiBoy screen (640x480).

duduke, 2021-09-08:
@dmrlawson 
Ok, I've had a quick go at the latest nightly build. 
First, it works out-of-the-box! :)

There are however several things that are not working right:
- The biggest issue I can see, is that for some reason, when using the nightly build, the lightning bolt symbol is constantly ON. This (I believe) leads to a significant reduction in speed. I've tried my image and the nightly build back-to-back to verify that my image doesn't display the lightning symbol. I've also compared several arcade roms and they were running slower on the nightly build (FBANeo core). I've also verified that the battery is fully charged.
- The menu defaults to RGUI - but that's a matter of taste. Ozone looks great on the PiBoy screen.
- Flipping the power switch initiates a safe shutdown. However, Choosing Shutdown from the RetroArch menu shuts down the OS, but does not signal the PiBoy to shutdown.
I think that "xpi_gamecon_shutdown.service" should be set to trigger when the OS is shutting down, this tells the xpi_gamecon driver to shutdown.

Let me know what I can do to assist, thanks.

dmrlawson, 2021-09-08:
> @dmrlawson
> Ok, I've had a quick go at the latest nightly build.
> First, it works out-of-the-box! :)
>
> There are however several things that are not working right:
>
> * The biggest issue I can see, is [...]

Thank you very much for testing :)

 - Lightning bolt: I haven't seen this at all. I just tested a fresh flash of the 2021-09-08 nightly and could play a psp game with no issues. Are you able to ssh in and check which version of the eeprom firmware you have on your pi? We've seen some odd issues on regular builds with certain firmwares.
 - I'm open to input on the menu. I had ozone with 150% and it seemed ok, but I think some users may find the text a bit small still. I chose to change it because then it matches what we do for other handhelds
 - Shutdown via menu I noticed too. I also found that using the switch while plugged in to usb power causes an unsafe shutdown, do you also see that?

For now I think it would be good to update the `xpi_gamecon_shutdown.service` as you say to fix these shutdown issues. I'm going to have a go when I find some time but if you figure it out feel free to make a PR or post what you think needs changing here and I'll add it.

duduke, 2021-09-09:
I've updated both the ExperimentalPi firmware to 1.0.5 and the Pi4 firmware.
I'm still getting the lightning bolt...
I think i've found another issue though:
After i've updated the ExperimentalPi to 1.0.5, the power switch would work incorrectly, it would initiate a shutdown, but the red LED would keep flashing. To fix that, I've added a `sleep 2` to my `piboy-dmg-control.sh` script (add this between lines 19-20).

dmrlawson, 2021-10-16:
@duduke thanks for that fix, I'll get it added in shortly.

@acidtech I've noticed an issue where the piboy appears to turn off during an update which can result in a non-working system. I think this might be because we're not loading the driver when we run the update script or something? Can you offer any advice or suggestions on how to prevent this happening?

acidtech, 2021-10-16:
If the Piboy is turned on and doesnt start communicating with the  RPi
board it will shutdown in 5 minutes IIRC.  Similarly there is a shorter
timeout if communications starts and then stops for length of time.

I assume when you say "update" you mean a lakka update.  Can you give me an
example of how this update is processed?

There is a special flag that can be sent to prevent a shutdown if
communications stops.  This is used then a Pi Reset is required since it
can take a while for the Pi to reset before starting the kernel driver
again.  I think this is likely to fix your problem?  See the
xpi_gamecon_reboot.service for an example.

/sys/kernel/xpi_gamecon/flags  holds the special flags.  Bit 0
enabled/disables the LCD.  Mainly used to turn off the LCD when using
HDMI.  Setting Bit 7 disabled the shutdown timeout.  For example writing
"129" to the file disabled auto shutdown and enables the LCD.

Note the kernel module must be loaded and active for this to work.

Nathan Scherdin

acidtech, 2021-10-16:
> @acidtech <https://github.com/acidtech> we compiled the module directly
> into the kernel to work around the issue with the system shutting down
> during update. this means that rmmod is not possible now. is that in
> issue?

That shouldn't be a problem. I can't think of a case you would need to
remove the kernel module outside of development purposes.

Nathan Scherdin

closed 2021-10-17 by ToKe79

dmrlawson, 2023-02-07:
@acidtech is it literally the same hardware and driver with extra axes? Might it just require a new controller profile, or do we need a new driver to pick up the extra stick?

Have you tested a recent Lakka DMG image on the XRS?

dmrlawson, 2023-02-08:
@acidtech I've messaged you on discord