libretro/Lakka-LibreELEC #1063: Update for Raspberry Pi 4 8GB

SIR-unit, 2020-10-15:
I downloaded Raspberry Pi 4 version of Lakka from the website.  However, it seems that the image provided is from March 2020, and there was a kernel/firmware update to the Pi 4 in May 2020.  I'm using the 8GB version of the Pi 4 which requires this update.  I have found others with the same problem, but it doesn't seem like anyone has submitted this as an issue here.

I have run other non-Lakka images with no problem.  However, when I boot the most recent Lakka version from the website, I get a screen asking me to go to raspberrypi.org/download to install new software.

Ntemis, 2020-10-20:
Please try this image
https://nightly.builds.lakka.tv/2020-10-20/RPi4.arm/Lakka-RPi4.arm-nightly-20201020-324ceda.img.gz

SIR-unit, 2020-10-20:
update: I ran downloaded an ran cave story with no problem, in case you wanted more confirmation