libretro/Lakka-LibreELEC #1067: WiFi broken in Lakka-H3.arm-3.0-devel-*-orangepi-plus2e nightly builds

cjom, 2020-10-20:
- Which version of Lakka are you using?

Lakka-H3.arm-3.0-devel-20200828000658-d9ba9d8-orangepi-plus2e.img.gz works very well.
All builds after that date have wifi broken.

- What system hardware are you using?

Capcom Home Arcade console uses a customized orangepi-plus2e.
Lakka-H3.arm-3.0-devel-20200828000658-d9ba9d8-orangepi-plus2e.img.gz works very well.

- What did you do?

Scan wifi to connect to my home network.

- What did you expect to happen?

To see a list of networks and connect to mine.

- What happened instead?

Builds from September: Lakka crashes and reboots; Builds from October: don't crash but no networks are found, showing only 1 garbage name (see picture).
[image]

I can't attach any logs because I'm unable to use SSH since it doesn't connects to wifi.

Ntemis, 2020-10-20:
Can you confirm that the onboard wifi chip is the Realtek RTL8189ETV?

cjom, 2020-10-20:
I see that it's not... it's RTL8188ETV (previous version?)
[image]

At least the build from 2020-08-28 works... but would be awesome if future versions would work too.

Ntemis, 2020-10-20:
That changes everything :)

cjom, 2020-10-20:
Any clue as why older builds work?
Or any tip about another version to test?
It has Allwinner H3 cpu and  for sure.

Ntemis, 2020-10-20:
https://github.com/libretro/Lakka-LibreELEC/commit/09408f8d668919b2ef95af73aef3740668ae5bc0

Ntemis, 2020-10-20:
try next nightly version and report back(you have to wait 24h)

cjom, 2020-10-20:
Thank you! Will test for sure!

Ntemis, 2020-10-20:
If Debian wiki doesn't lie it should work fine :)
https://wiki.debian.org/rtl819x#supported-r8188eu

ToKe79, 2020-10-21:
New image is up - [Latest images - Allwinner H3](https://nightly.builds.lakka.tv/latest/H3.arm/)

cjom, 2020-10-21:
And it works!
Thank you so much!

closed 2020-10-21 by cjom