libretro/Lakka-LibreELEC #1072: Possible RG351p support? [enhancement]

iokzero, 2020-10-24:
Not sure if this is the right place to post this, please remove or relocated as needed. 

The RG351p is a rockchip RK3326 device that is very similar to the Odroid Go Advance (OGA) with the same CPU and pretty much everything else, except it comes with a different built-in controller with rumble support. 

The current OGA nightly build seems working fine with the device, except the rumble motor would stay on as soon as the device got turn on. and the controller(s) doesn't seem like doing anything. Even when external gamepad is connected. Retroarch will see the external gamepad correctly and is show on the notification bar, but no button does anything. 

https://github.com/lualiliu/RG351P-linux

found a fork for the kernel that seems targeting on issues for this specific device. not sure if it is any useful. seem like only the controller and battery values need some adjustment. 

Would love to help testing if developers are interested to bring in support. I really like the device, great build quality and button layout. 

Many thanks!

Ntemis, 2020-10-27:
I dont know anyone with that device and if we dont have it we cant work on it

davidgfnet, 2020-10-27:
It's the same SoC right? RK3326?
I guess it's just a matter of rewiring the GPIOs for the controller/rumble. Does the other things work? Well without a gamepad I guess you can't test it right? :D
Any working distro that we can steal the GPIO mapping from?

iokzero, 2020-10-28:
It is using the same RK3326 CPU, and seem like everything else too. 

Emuelec works, since 3.6 (it is 3.9 current). But I think it is custom build or something. The controller would work under official Emuelec, but the rumble motor would still goes non-stop. 

Not sure if you can figure it out from a built image, but here's the link
https://drive.google.com/file/d/1RX66hUqSlJgNxk6z7Te8cOdBvEiEMjbf/view

They also released a controller firmware patch while back
https://drive.google.com/file/d/1E--wv1cul6zXSRXhxgtiqm6u6Mmom6T-/view

Again, not sure if you can figure out the GPIO mapping from there either. 

Can help with testing if that help, 

Many thanks

iokzero, 2020-10-28:
Actually, Just discover that 

batocera just released a beta for this, and got it tested, seems working fine so far. 
https://github.com/batocera-linux/batocera.linux

I found out from this page:
https://github.com/retrogamehandheld/oga/wiki/Custom-Firmwares

Again thanks for the help!

davidgfnet, 2020-10-28:
Batocera is using the same Go Advance image, with a couple of tweaks for this platform. Shouldnt be too hard to add them. IIUC we already support Go Advance v1 and v1.1 right? We could add this platform too.
We would need someone to test it though

iokzero, 2020-10-29:
I have the device handy, and can help with testing if there's a built image. or if there's another ways I can help, please let me know.

iokzero, 2020-11-16:
https://github.com/fewtarius/351ELEC

Another good reference build for this specific device.

davidgfnet, 2020-11-18:
Sent PR #1105

iokzero, 2020-11-19:
Thank you so much for effort and support!

davidgfnet, 2020-11-19:
The nightlies are out. Feel free to test and report any bugs. You can ping me on Discord if it's easier.

We can close this, it's fixed.

closed 2020-11-19 by Ntemis

iokzero, 2020-11-20:
Thank you so much for the work!. 

Just did a quick test. And here are some findings 

What works:
The controller works 
The rumble motor no longer go crazy. 
At least one core for these systems works:
NES, SNES, GB, GBC, GBA, WonderSwam/Color, PSX, PSP, N64, PC-Engine/CD, Mr.Boom,  

What doesn't works:
Battery indicator doesn't seem showing the correct percentage. 
FlyCast wouldn't launch. 
Mupen64-next works, but Mupen64-parallel doesn't. 
Ozone works, but it is way too small to be useful, You can scale it up to about 1.7X to make it slightly better, but it wouldn't scale up from there. 
No way to test the rumble function, but already assumed it wouldn't work at the moment. 

Overall I think Lakka works really well on the RG351p, and most of problem doesn't seem to be specific to RG351p, but more general OGA/OGA clone issues in general. Except the Battery indicator (I think.)

Will do some more in dept testing soon.

@davidgfnet  I am not a Discord user in general, but would love to join if that makes things easier. Do Lakka has its own Discord?

vanfanel, 2020-11-21:
@davidgfnet Thanks for adding support for this device, I recently got one and I have just tested nightly Lakka on it, which boots fine as far as I can see.

However, performance is VERY poor: for example, it can't even run Akumajo Densetsu (Castlevania III Japanese version) on the Nestopia core at full speed.
Official firmware runs this game with no difficulties at all using the exact same core. This is a very visible example, but you can see occasional hiccups in any NES game in Nestopia, in MegaDrive games in Genesis Plus GX core, etc. Those are not present in original EmuELEC-based firmware.
Same hiccups can be seen in 351elec, etc. 

So I believe there's something different in official firmware, deeply affecting performance. I am reporting this here because RetroArch community seems the only sensible to these problems.

vanfanel, 2020-11-21:
@iokzero I am seeing **MegaDrive games** in **Genesis Plus GX** fail just after booting (they cause the core to fail and RetroArch menu is loaded), using latest Lakka nightly. Can you reproduce? Trying to Play Sonic 1 is enough.

iokzero, 2020-11-21:
@vanfanel Yes, I can reproduce the issues you mentioned. In fact, most of my game will run, but it will crash everything once you are in title screen for that core. 

@davidgfnet Do you think we should separate the issues from RG351p and generic RK3326 device issues?