libretro/Lakka-LibreELEC #1077: cubieboard2 nightly [KMS]: Couldn't find a suitable DRM device.

johndoe71rus, 2020-10-30:
- Which version of Lakka are you using? Lakka-H3.arm-nightly-20201025-97f7975-cubieboard2.img.gz
- What system hardware are you using? cubieboard 2 Allwinner A20
- What did you do? try to boot lakka
- What did you expect to happen? successfully boot
- What happened instead? stuck at start Retroarch

log retroarch -v
https://paste.ubuntu.com/p/S7f6WPt65v/
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault

by the way, nightly don't have any .dts files. I take it from lakka cubieboard2 image at 24 jan 2020

gouchi, 2020-11-06:
Thanks to the work of @zhangn1985 from #1083 we have some update for Allwinner.

It will need some adjustments but it is a good start.

You can make some test with the [new nightly](https://nightly.builds.lakka.tv/latest/H3.arm/).

Thank you.

Ntemis, 2020-11-06:
A20 is abandoned even upstream
https://github.com/LibreELEC/LibreELEC.tv/commit/cf5e056881caf9583f9ce63155e71f6170b918bd#diff-d04d94da639bf30ad2ea549c8dd3c92ff7db4b1fd9b9f22bbc5e54610a3bae09

closed 2020-11-06 by Ntemis

johndoe71rus, 2020-11-07:
@gouchi
The result is the same.
For me nightly libreelec builds worked well. Except for the hdmi audio broken in February. YouTube played well.
Libreelec dropped the A20 because no one was interested in the topic.
If @zhangn1985 is working on this topic, I would like to contact him. It is a pity that there are no private messages on github.
By the way, lakka and libreelec use different sun7i-a20-cubieboard2.dtb files. In lakka it is 2 times more.