libretro/Lakka-LibreELEC #1082: Lakka does not work on Intel NUC Hades Canyon

ChojinDSL, 2020-11-04:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
Lakka-Generic.x86_64-2.3.2.img

- What system hardware are you using?
Intel NUC Hades Canyon NUC8i7HVK

- What did you do?
Run the live USB

- What did you expect to happen?
See the installation screen

- What happened instead?
Black screen.

The Intel NUC Hades Canyon is a bit of a unique situation, since it actually has two GPUs. There is the onboard intel as part of the Core I chipset, but also an AMD Radeon RX Vega M Series. The onboard intel gpu is actually not used. As far as I can tell it's not even routed to any of the video outputs. 

Standard linux distros boot just fine on this hardware. I suspect that due to the minimalist architecture of lakka, that it tries to output via the onboard intel GPU, which of course results in a blank screen.

Is there any way that I can perhaps force it to use the AMD gpu instead?

Mr-Jake, 2020-11-04:
You need to use Lakka 3.0 Nightly.

https://nightly.builds.lakka.tv/

ChojinDSL, 2020-11-09:
Ah, ok. Thanks.

closed 2020-11-09 by ChojinDSL