libretro/Lakka-LibreELEC #1084: Does not seem responsive on the Rpi4

abnormisdictum, 2020-11-06:
I burnt lakka on a SD card for my Rpi4 4GB. After the initial bootup resizing, i was able to connect to my wifi network. But when I tried to update Lakka, it simply stated that task failed and that there were no entries to display.

So i tried loading a psp core. But after selecting the core nothing happened.

Mind you i was using a keyboard and not a controller. Could that be the issue?

I would appretiate any help on this to point me in the right direction.

Warm regards.

Ntemis, 2020-11-06:
What lakka version did you use?

abnormisdictum, 2020-11-06:
2.3.2

abnormisdictum, 2020-11-06:
> 2.3.2. The latest one available on the website lakka.tv

Ok. To be certain it's not because of a lack of a controller, i used my XBox 360 wireless controller.

The issue persists

Ntemis, 2020-11-06:
Try migrating to a nightly version

abnormisdictum, 2020-11-06:
> Try migrating to a nightly version

Will give it a go and report back.

Ntemis, 2020-11-12:
Please switch to more recent nightlies versions from here https://nightly.builds.lakka.tv/
Version 2.3.2 is the last version on v2.x branch and is not destined to get any more updates or bug fixes.
Switching to nightlies help us too to catch and fix bugs before the final release and prevents any issues like this show up in the future.

closed 2020-11-12 by Ntemis