libretro/Lakka-LibreELEC #1085: Playstion psx/ps1 image distortion

Rstation4, 2020-11-07:
Raspberry Pi 4 model B 
Lakka stable image 2.3.2

I'm getting these image distortions from the following: Glover, Monkey Hero and Guardian's Crusade. I'm not sure if its a ROM issue or a core issue. This is from using the only Playstion core that is used currently on the Raspberry pi 4b lakka 2.3.2 image. One of them uses a ccd format.

![Glover (USA)-200120-184445](https://user-images.githubusercontent.com/73009731/98427099-9e492b80-2050-11eb-98c9-266e434
[image]
311f8.png)
[image]

Ntemis, 2020-11-07:
https://nightly.builds.lakka.tv/2020-11-05/RPi4.arm/

Ntemis, 2020-11-12:
Please switch to more recent nightlies versions from here https://nightly.builds.lakka.tv/
Version 2.3.2 is the last version on v2.x branch and is not destined to get any more updates or bug fixes.
Switching to nightlies help us too to catch and fix bugs before the final release and prevents any issues like this show up in the future.

closed 2020-11-12 by Ntemis