libretro/Lakka-LibreELEC #1086: 8bitdo  SN30pro wired controller support RPI4b Lakka stable release 2.3.2

Rstation4, 2020-11-07:
Raspberry Pi 4 model B, Lakka image 2.3.2 stable

It gets detected as a xbox 360 gamepad, while I have not tested all the cores with it, it does have input errors on the ps1 core and sometimes becomes unresponsive with that core.

Ntemis, 2020-11-07:
https://nightly.builds.lakka.tv/2020-11-05/RPi4.arm/

Rstation4, 2020-11-10:
I used the nightly for 11-05-2020 and it still gets detected as a xbox 360 controller, If i try the ps1 cores again I will update this post than to drag out multiple posts and clutter this issue page.

Ntemis, 2020-11-12:
Please switch to more recent nightlies versions from here https://nightly.builds.lakka.tv/
Version 2.3.2 is the last version on v2.x branch and is not destined to get any more updates or bug fixes.
Switching to nightlies help us too to catch and fix bugs before the final release and prevents any issues like this show up in the future.

closed 2020-11-12 by Ntemis