libretro/Lakka-LibreELEC #1088: low frame rate and core crashing N64 Rpi4b

Rstation4, 2020-11-07:
Raspberry pi 4 model B
Lakka stable 2.3.2

When using Mupen64plus it has low frame rates I only tested it out on 3 roms, Paper Mario, Mario Kart (had some pixelated lines) and Kirby 64. Trying the other core Parallel ended up crashing Lakka. The settings on those 2 cores has not been touch and they are at the default settings.

Cibomatto2002, 2020-11-07:
Try Lakka - Nightly see if that helps https://nightly.builds.lakka.tv/2020-11-05/

Rstation4, 2020-11-10:
I used the nightly for 11-05-2020 and it did not solve it. I may not test another nightly anytime soon.

Ntemis, 2020-11-10:
Try to lower the resolution. RPi4's gpu is not much more powerful than the rpi3+ one so don't expect miracles.
 Parallel  needs a gpu that conforms to ogles v3.2 but Rpi4 gpu only offers v3.1, i wouldn't call this a bug per se but we will investigate if we can fix it never the less.
Thanks you for reporting this.

Ntemis, 2020-11-12:
Please switch to more recent nightlies versions from here https://nightly.builds.lakka.tv/
Version 2.3.2 is the last version on v2.x branch and is not destined to get any more updates or bug fixes.
Switching to nightlies help us too to catch and fix bugs before the final release and prevents any issues like this show up in the future.

closed 2020-11-12 by Ntemis