libretro/Lakka-LibreELEC #1089: Won't boot up on RPi 4 B - Fresh Install

graydon2212, 2020-11-08:
I have a brand new Raspberry Pi 4 B.  I previously set it up to work with Raspberry OS and everything went well.  I also flashed RetroPie to an SD Card without trouble.

I just tried to do the same with Lakka.  Downloaded the Lakka-RPi2.RPi4.arm-2.3.2.img file, etched it with balenaEtcher. Put it in the Raspberry pi, and it fails to boot, giving this error (see photo)
[image]

it seem to be looking for start4.elf, which does not appear on the freshly etched SD Card (See image)

[image]

Any ideas?  Am I getting something wrong here?

This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

Cibomatto2002, 2020-11-08:
try this nightly https://nightly.builds.lakka.tv/2020-11-05/

Ntemis, 2020-11-10:
2.x images dont support recent RPI4 versions so stick to the nightlies please

closed 2020-11-10 by Ntemis