libretro/Lakka-LibreELEC #1095: General issues with lakka

Enrico-Fanelli-97, 2020-11-10:
Since I wasted a day and half on working around this distro meanwhile by installing Windows 7 with RetroArch I had everything up and running within an hour, I felt the need to report to someone the god awful experience I had and maybe someone might be able to fix such problems. The version I downloaded is the 2.3.2 on x86 recycled computer

- Intel Pentium E2200
- 3GB DDR2 667 MHz
- Asus IP4BL-ME (OEM only motherboard)
- ATI HD5870
- CN WF551 / Asus WL-138G V2 (Wireless network adapter)

The pressing issue is the Wifi adapter by itself. At first I thought that the Asus WL-138G V2 was simply unsupported since I kept receiving gibberish as access points, so I fall back onto the CN WF551 which I used a long time in another linux machine and know it works. At first it picked up all the networks and I was able to log into my access point, however it broke again at the subsequent reboot. Same problem: the network scan just returns a single "access point" which is pure gibberish. As I've read I'm not the only one to have encountered such problem.

This wouldn't have been much of a problem if I were able to access the terminal, however some smart guy thought it was a smart idea to ship Lakka with the terminal disabled and, surprise surprise, the only way to enable it is through SSH. The only way to access the partition is by physically remove the hard drive and use another computer.

As this is annoying by itself, the final problem I had with the large partition itself. Why can't I choose the filesystem during the installation like everyother linux distribution out there? Why is it all locked with ext4? Windows 10 has quite some problems with ext4 drivers and I was completely unable to even see the what's on the disk. This has been all but an _out of the box experience_ and I hope someone put some remedy to this. This is more of a feedback than actual bug report.

Ntemis, 2020-11-10:
Please try to switch a a nightly image from here https://nightly.builds.lakka.tv/

Ntemis, 2020-11-12:
Please switch to more recent nightlies versions from here https://nightly.builds.lakka.tv/
Version 2.3.2 is the last version on v2.x branch and is not destined to get any more updates or bug fixes.
Switching to nightlies help us too to catch and fix bugs before the final release and prevents any issues like this show up in the future.

closed 2020-11-12 by Ntemis