libretro/Lakka-LibreELEC #1101: Outdated docs

tavuntu, 2020-11-16:
I'm trying to compile the last version of lakka (currently 2.3.2) and I had no mayor issues, but [this page](https://www.lakka.tv/doc/Compiling-Lakka/) seems to be outdated (the script asks me to set the DEVICE and PROJECT values). I'm compiling for Raspberry Pi.

So, in short, this doesn't work:
```DISTRO=Lakka PROJECT=RPi2 ARCH=arm make image```

This did work.
```DISTRO=Lakka PROJECT=RPi ARCH=arm DEVICE=RPi make image```

Also, this may be a separate issue but since it's related I'll ask here as well. Is there no RPi3 device for the RPi project? I get this list when no DEVICE value is set in the make command:

```
Valid devices for project: RPi
 - GPICase
 - Gamegirl
 - RPi
 - RPi2
 - RPi4
 - Slice
 - Slice3
```

Rstation4, 2020-11-22:
I second this, the docs are outdated.

SethBarberee, 2020-11-23:
RPI2/3 are build together. Image supports both. I know that Lakka was moving over to libretro's docs and I've already submitted a PR to update that set of docs.

https://github.com/libretro/docs/pull/560

ToKe79, 2021-06-10:
docs were updated, but bear in mind that they serve as an example, not step by step guide

closed 2021-06-10 by ToKe79