libretro/Lakka-LibreELEC #1111: [OGA][RG351] Remaining issues

vanfanel, 2020-11-22:
Hi,

I recently received my RG351P, and of course Lakka became my instant choice as OS to use on it.
I found that OGA/RG351 was just added by @davidgfnet , first nightly version including it is here:

https://nightly.builds.lakka.tv/2020-11-19/OdroidGoAdvance.arm/

However, there are some small issues with it:

-GenesisPlusGX: MegaDrive games fail after some seconds, going back to the RetroArch menu. Master System/GameGear games don't do this. 
 Seems like core-related to me: build options, included core version, etc.

-Battery charge is not correctly displayed, nor is LED indicator correctly set up to green on charged unit.

-Hybernation doesn't work: it usually works on a light push on the power button.

None of those issues happen in 351elec (Emuelec fixed for RG351) available here:
https://github.com/fewtarius/351ELEC
...So maybe the charge indicator are related to kernel/dtb?

jdalmanza, 2021-01-10:
Tried nightly from 01/08 and can confirm the GenesisPlusGX behavior exists on OGA BE as well.

THTB614, 2021-01-16:
From a bit of digging, it looks like the battery display is essentially reading the battery as if it were the OGA's battery, which is 500MaH smaller in capacity, if that helps any.

Ntemis, 2021-04-06:
@davidgfnet can we close this now?

jdalmanza, 2021-04-07:
just tested nightly from 04/06 and genesis-plus-gx still crashes when gameplay starts

ToKe79, 2021-04-09:
#1248 should have fixed the genesis crash.

davidgfnet, 2021-04-11:
Yep, genesis issue is now fixed! Seems like aarch64 kernels are not 100% compatible with armv7 binaries :) For the curious: https://news.ycombinator.com/item?id=18855047
I'm gonna tacke the battery and sleep mode things soon, I've got some work in progress stuff sitting on my local git.

davidgfnet, 2021-04-14:
Sent PR #1262 for the battery issue

closed 2021-07-26 by Ntemis