libretro/Lakka-LibreELEC #1112: [Feature Request] scripts for safe shutdown buttons and reset buttons

Rstation4, 2020-11-25:
Raspberry Pi 4 B
Lakka OS: Nightly Nov 19,2020 rpi4 arm

SethBarberee, 2020-11-25:
This isn't really something to add to the default image. However, there are ways to do it with GPIO overlays or other programs. I believe a few of the other projects in the repo for commercial products have safe shutdown scripts.

One example:
https://github.com/edus44/lakka-gpio-control/tree/master/scripts

Rstation4, 2020-11-25:
Is there something that has changed for the shutdown process on the current images for the rpi4b? I have a NesPi 4 case that has not worked with either sripts.
 https://github.com/marcelonovaes/lakka_nespi_power   That is the one I have used before without an success

Ntemis, 2021-07-28:
Not happening unless we have the case for r&d case closed

closed 2021-07-28 by Ntemis