libretro/Lakka-LibreELEC #1113: Did not boot on Raspberry Pi4 2GB

matzeeg3, 2020-11-25:
I flash the Lakka image to a Micro SD Card but than i only get:
start.elf: is not compatible 0x0
This board requires newer software.

is this a know issue? how can i fix it?

SethBarberee, 2020-11-25:
Some more info is needed here. Which image did you download?

matzeeg3, 2020-11-25:
Sure I use this image:
http://le.builds.lakka.tv/RPi2.RPi4.arm/Lakka-RPi2.RPi4.arm-2.3.2.img.gz
And the raspberry pi 4 is the normal 2gb Version. It looks like the latest firmware is installed. I just brought ir

SethBarberee, 2020-11-25:
Try the nighties and see if your situation improves. There have been a lot of work on RPi4 since 2.3.2

https://nightly.builds.lakka.tv/

matzeeg3, 2020-11-26:
i will try?
do i need rpi4.arm or rpi4.aarch64?
is there a next release date?

matzeeg3, 2020-11-26:
so now i see lakka is booting useing image: 
Lakka-RPi4.arm-nightly-20201126-96c42d3.img.gz

matzeeg3, 2020-11-28:
And with the new image I am note able to use save shutdown. I get the message that the rpi library is only for rpi.

Ntemis, 2021-06-06:
Please test 3.1 release