libretro/Lakka-LibreELEC #1115: No video output to 4K TV, but works with hdmi monitor

graydon2212, 2020-11-30:
I have installed Lakka on my Pi 4b 8GB.  Works fine with a monitor, but will not provide a video output to my Samsung TV.  Boot the same Pi with another add card running Rapberry OS everything is fine.  Must be a software issue. 

This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

Ntemis, 2020-11-30:
Try a nightly build please

graydon2212, 2020-12-01:
Hi,  I'm already using a nightly build.  On the Lakka.TV website if you choose a Pi 4 B as your device, it points you to a Pi 3 installation, which does not work!  I was advised to use a nightly version to resolve this.

Cibomatto2002, 2020-12-03:
https://nightly.builds.lakka.tv/2020-12-03/

relaxxxrrr, 2020-12-04:
I have similar problem with Orange Pi Lakka 2.3.1. No video signal on two of my HDTV's. Then I tried a 4K TV and it shows a very small #lakka prompt in 4K resolution. It doesn't boot past that. I had an older version 2.0 that booted fine. I can't seem to find any older versions to test anywhere??

Cibomatto2002, 2020-12-04:
> 
> 
> I have similar problem with Orange Pi Lakka 2.3.1. No video signal on two of my HDTV's. Then I tried a 4K TV and it shows a very small #lakka prompt in 4K resolution. It doesn't boot past that. I had an older version 2.0 that booted fine. I can't seem to find any older versions to test anywhere??

 If you can change the TV screen resolution set it to something like 1080 or 720 see if that helps.

relaxxxrrr, 2020-12-04:
Sorry, I have no idea how. Like a boot config file? All I see are a bunch of DTB files and a uEnv.txt file. Yeah, I'm completely lost.

Cibomatto2002, 2020-12-04:
see if this helps if not i have no idea.

https://forums.libretro.com/t/manually-configuring-resolution/6323

closed 2021-07-15 by ToKe79