libretro/Lakka-LibreELEC #1118: Playlists Black Icons

GitCoder23, 2020-12-02:
I'm Using A Raspberry Pi Zero W.
And Everything Usually Works Fine.
And The Latest Build Of Lakka.

In The Menu Playlists , Instead Of Icons , It Is Black.

SethBarberee, 2020-12-02:
> And The Latest Build Of Lakka.

Not much help here. Which exact image did you download? From the website? A nightly (if so, which date)?

GitCoder23, 2020-12-02:
From https://www.lakka.com/get

A Standard Image , Not a Nightly

I Will Try And Provide Images

SethBarberee, 2020-12-02:
Try a nightly from:
https://nightly.builds.lakka.tv/

`RPi.arm` most recent date and see if your problem is fixed. 2.3.2 is an older release.

GitCoder23, 2020-12-03:
Ok

jdalmanza, 2021-02-21:
noticed the same issue on my rpi3 while curating my playlists. I found that if the thumbnail image is too large (larger than 1920x1080) it does not render the image and you get a "black" thumbnail. Same is true for graphical overlays. This is present on the most recent nightly builds last tested on devel-20210213015453-8076be1

gouchi, 2021-02-21:
We already have an [issue](1046)  for the thumbnails turning black.

jdalmanza, 2021-02-21:
> We already have an [issue](1046) for the thumbnails turning black.

I was just offering my experience, it took me a bit to troubleshoot this one :-D

closed 2021-07-13 by ToKe79