libretro/Lakka-LibreELEC #1128: PS1/PSX  PCSX ReARMed core bugs

Rstation4, 2020-12-10:
Raspberry Pi 4 model B
Nightly Build Lakka RPI4 ARM Nov-9-2020

With some ROM's there are some compatibility errors. 

Legend of Mana: I've tried several rom versions and for some reasons the controller inputs do not work its a weird bug to explain.

Brave Fencer Musashi: screen went black and stayed that way until reseting, in past **Lakkka** versions there has been slowdowns as well.

Quake: upon reaching a certain level I get graphical errors

Final Doom: graphical glitches so far when using the higher resolution option.

[image]
[image]
[image]
[image]

Ntemis, 2020-12-10:
can you please retest using this image?
http://demetris@buildserver3.lakka.tv:8080/Lakka-RPi4.aarch64-3.0-devel-20201210183618-f79b1b0.img.gz
Thank you in advance

Ntemis, 2021-06-06:
Please test 3.1 release

closed 2021-07-23 by Ntemis