libretro/Lakka-LibreELEC #1141: Easyrpg core fails to run, libicu.so is missing from image

asdf288, 2020-12-20:
- Which version of Lakka are you using? Lakka master armhf
- What system hardware are you using? Amlogic S905x
- What did you do? Build and install the Lakka image, run the easyrpg core
- What did you expect to happen? Core runs
- What happened instead?

The easyrpg core fails to run with the message "Failed to load Libretro core". Running Retroarch with the --verbose flag reveals that the core loads liblcf.so, which tries to load the libicu.so* libraries that are not present on the image. Installing these libraries (using the fix described below) reveals another error: libicu is found, but liblcf.so again fails to load because it attempts to use missing symbols from libicu.so.

To fix these errors:

First, fix the missing libicu.so files. This issue is caused by the [icu package](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/addons/addon-depends/chrome-depends/icu/package.mk). It builds the shared libicu library for the target, installs them to the toolchain, but then, oddly, deletes them from the image in the last step (post_makeinstall_target). This means that they're missing from the final image.

**To fix this:** remove the post_makeinstall_target part from the package.mk entirely.

The next error happens when liblcf tries to load the icu library. This fails with a message `/usr/lib/liblcf.so.0: undefined symbol: _ZTVN6icu_6813UnicodeStringE`. Analyzing the libicu.so using `strings libicu.so | grep _ZTVN6icu_6813UnicodeStringE` indeed reveals that the symbols are missing. This apparently happens because liblcf is compiled with symbol renaming enabled, but the icu package has it disabled by default.

**To fix this:** Compile the icu package with symbol renaming enabled (add `--enable-renaming` in PKG_CONFIGURE_OPTS_TARGET on line 27).

Fixing these two issues made the easyrpg core load successfully on my image.
 
Regarding the first error, the original intent was probably to build icu as a shared library, link the programs against it and then delete it because the static libraries are not needed in the image anymore. I tried to build icu and liblcf as static libraries, but the easyrpg core then fails to build with a message that a functions from libicu cannot be relocated and the library needs to be recompiled using -fPIC. Doing this had no effect. Googling the issue gave a hint that this might be resolved by compiling icu without assembly, but I could not get this to work and it exceeds my limited knowledge, so I gave up at this point.

Please tell me if you can reproduce this error and if the above steps fix it for you.

Ntemis, 2020-12-20:
Thank you for the time invested in this.
Can you please send a pr with the changes?

asdf288, 2020-12-20:
Yes, I will do that. It could take a bit of time though, I have never created a pull request before, I have very little experience with git and github.

Another thing to consider: I have no idea if the second fix (--enable-renaming) could potentially break other things. Apparently, the only other package that makes use of icu right now is the chrome browser addon, which is from LibreELEC I think. But still, there could be a better way to solve this. Maybe it's possible to build libclf with renaming support disabled so that it works with the original icu package, I will give that a try.

Ntemis, 2020-12-20:
When you are ready. You can join lakka discord if you need further help
creating the pr. I will be more than happy to assist you in every step 😊

PR #1143 (2020-12-20): Fix easyrpg core runtime error (missing icu4c libraries)
Fixes #1141 
Add the missing ICU shared libraries and fixed the following bug:

ICU can be compiled with or without symbol renaming support. Apps need to consider this by including a switch in the compiler options. This information is stored in the icu pkgconfig file. Libclf does not check for this and assumes that symbol renaming is enabled. In the case of Lakka it is disabled though, leading to a runtime error where the libicu.so* symbols cannot be found due to a naming mismatch.
closed 2020-12-20 by Ntemis