libretro/Lakka-LibreELEC #1148: [Feature Request] Dolphin Wii Mote & Nunchuk Tilt and Calibrate Settings

TheBabigoon, 2020-12-26:
**Hello Anyone! 👋**
*maybe, i should request this features on retroarch. i'm sorry 😢
### when play any game with Tilt like this, that's kinda weird.
[image]

### on the official dolphin emulator there is Tilt and Calibrate setting.
**image for Wii Mote**
[image]
**image for Nunchuk**
[image]

### or maybe swing too, that's big help for keyboard and controller users.
**thanks for any help! 😄**

Ntemis, 2020-12-26:
Hi, not even retroarch.
This issue belongs here https://github.com/libretro/dolphin

RobLoach, 2020-12-26:
https://github.com/libretro/dolphin/issues/186

closed 2020-12-26 by RobLoach

TheBabigoon, 2020-12-28:
@RobLoach @Ntemis thanks