libretro/Lakka-LibreELEC #1150: Pine64+ Image - ethernet, wifi, bluetooth doesn't work

dewolft, 2020-12-28:
Hello, i've looking in the forum and on google, but there is no information about it,
i'm use a : Pine A64+ 2016 kickstarter version with 2gb of ram
Latest image : Lakka-A64.arm-nightly-20201221-62cd91f-pine64-plus.img
i've just flash into a blank 32gb SD Card
if i go to the System information and looking for the ethernet address, i'll get just the Interface(io) 127.0.0.1
if i try to scan Wifi, nothing happen
if i try to scan Bluetooth, nothing happen

the other functionality seems to works well ( i've tried to upload a game directly from the SdCard and it's worked )

thanks
Riccardo

gmello77, 2021-01-07:
Hello,
same problem for me.
i'm use a : Pine A64+ 2016 kickstarter version with 1gb of ram
Latest image : Lakka-A64.arm-nightly-20201221-62cd91f-pine64-plus.img
No wifi only LAN, I get Interface(io) 127.0.0.1
Audio through HDMI doesn't work.

Thanks

SethBarberee, 2021-01-09:
HDMI Audio isn't supported yet for any A64 device on linux. The driver is still WIP.

I don't recall which driver supports the Wifi/Bluetooth RTL8723BS but we might be missing the driver. Haven't done a lot of research into making it work on Lakka.

gouchi, 2021-05-02:
@dewolft @gmello77 May you try this [image](https://nightly.builds.lakka.tv/members/gouchi/1150/Lakka-A64.arm-3.0-devel-20210502140458-4685525-pine64-plus.img.gz) ?

Thank you.

gmello77, 2021-05-02:
@gouchi I tried it right now:

- LAN, I get Interface(io) 127.0.0.1 : not working
- Audio through HDMI not working.

my Pine doesn't have wifi or bluetooth so I can't try

Thank you for your effort

gouchi, 2021-05-02:
@gmello77 Thank you for the test may you give the output of `dmesg` and `lspci -nnk | grep -i -A3 ethernet`.

For the audio, I did not change anything but may you give the output of `aplay -L`.

If needed, the [documentation](http://www.lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

gmello77, 2021-05-08:
@gouchi Sorry I think I need a little help so can't access the terminal because neither ethernet nor wifi works. I tried direct access modifiyng extlinux.conf this way 
`LABEL Lakka
  LINUX /KERNEL
  FDT /sun50i-a64-pine64-plus.dtb
  APPEND boot=UUID=0205-0725 disk=UUID=cfddb7e6-c799-4856-badd-ffedf9260c85 quiet console=ttyS0,115200 console=tty1 retroarch=0`
  but after CTRL + ALT + F3 I can't use terminal because is stuck. Any suggestions?

gouchi, 2021-05-08:
@gmello77 May you try with `tty retroarch=0` or `textmode` ?

Thank you.

gmello77, 2021-05-10:
@gouchi I did it with `textmode` :)

`dmesg` returns

`[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.1.18 (gouchi@globox) (gcc version 8.3.0 (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36))) #1 SMP PREEMPT Sun May 2 14:05:18 CEST 2021
[    0.000000] Machine model: Pine64+
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 384 MiB at 0x0000000068000000
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x67ddc840-0x67dddfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 262144
[    0.000000]   DMA32 zone: 4096 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 262144 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.1
[    0.000000] random: get_random_bytes called from start_kernel+0x9c/0x44c with crng_init=0
[    0.000000] percpu: Embedded 23 pages/cpu s53528 r8192 d32488 u94208
[    0.000000] pcpu-alloc: s53528 r8192 d32488 u94208 alloc=23*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] Speculative Store Bypass Disable mitigation not required
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Policy zone: DMA32
[    0.000000] Kernel command line: boot=UUID=0205-0725 disk=UUID=cfddb7e6-c799-4856-badd-ffedf9260c85 quiet console=ttyS0,115200 console=tty1 retroarch=0 textmode
[    0.000000] Memory: 619152K/1048576K available (9276K kernel code, 772K rwdata, 3584K rodata, 3008K init, 349K bss, 36208K reserved, 393216K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=4.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: Enabling global workaround for Allwinner erratum UNKNOWN1
[    0.000000] arch_timer: CPU0: Trapping CNTVCT access
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000005] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000417] Console: colour dummy device 80x25
[    0.000428] printk: console [tty1] enabled
[    0.000491] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.000501] pid_max: default: 32768 minimum: 301
[    0.000600] LSM: Security Framework initializing
[    0.001215] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.001514] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.001550] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.001566] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.002096] *** VALIDATE proc ***
[    0.002369] *** VALIDATE cgroup1 ***
[    0.002376] *** VALIDATE cgroup2 ***
[    0.024026] ASID allocator initialised with 32768 entries
[    0.032024] rcu: Hierarchical SRCU implementation.
[    0.040857] EFI services will not be available.
[    0.048062] smp: Bringing up secondary CPUs ...
[... 285 lines cut ...]
`aplay -L` returns
`null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=sun50ia64audio
    sun50i-a64-audio, 
    Default Audio Device
sysdefault:CARD=sun50ia64audio
    sun50i-a64-audio, 
    Default Audio Device
front:CARD=sun50ia64audio,DEV=0
    sun50i-a64-audio, 
    Front output / input
default:CARD=allwinnerhdmi
    allwinner-hdmi, 
    Default Audio Device
sysdefault:CARD=allwinnerhdmi
    allwinner-hdmi, 
    Default Audio Device
hdmi:CARD=allwinnerhdmi,DEV=0
    allwinner-hdmi, 
    HDMI Audio Output
`

hope this helps.
Thank you!

gouchi, 2021-05-14:
@gmello77 Thank you for the log.

It seems the ethernet is detected :

```
[ 7.551386] RTL8211E Gigabit Ethernet stmmac-0:01: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=stmmac-0:01, irq=POLL)
...
[ 11.652457] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[ 11.652500] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

I forgot to ask you the ouput of `lsmod` and `grep REALTEK /lib/modules/$(uname -r)/build/.config`

For the audio you tried all this device audio : `default:CARD=allwinnerhdmi`, `sysdefault:CARD=allwinnerhdmi`, ` hdmi:CARD=allwinnerhdmi,DEV=0` and it is not working for the HDMI ?

Thank you.

gmello77, 2021-05-17:
@gouchi 
`lsmod` returns:
Module                   Size              Used by
crct10dif_ce           16384          1
dwmac_sun8i         24576         0
mali                         370200       0
ipv6                        438272        32

`grep` returns:
grep: /lib/module/5.1.18/build/.config: no such file or directory

For the audio: I manually added device in audio as you said and it works! 

Thank you

gouchi, 2021-05-18:
@gmello77 thank you for the information. I am glad it is working for the audio.

Your LAN is only IPv6 not IPv4 ?

Thank you.

gmello77, 2021-05-19:
@gouchi my LAN is in IPv4 too. If I install dietpi (https://dietpi.com/) on Pine64, LAN is correctly detected in IPv4

Ntemis, 2021-05-19:
Since we are based on libreelec can you please try their image and let us know? Thank you

gmello77, 2021-05-21:
@Ntemis I downloaded librelec `LibreELEC-A64.arm-10.0-nightly-20210519-f9d2c33-pine64-plus.img.gz` from here https://test.libreelec.tv/, audio and LAN worked immediately out of the box (watched an episode of south park in streaming btw, worked like a charm).
Thank you

gouchi, 2021-05-21:
This is the patch I applied for the testing image 

```
diff --git a/projects/Allwinner/linux/linux.arm.conf b/projects/Allwinner/linux/linux.arm.conf
index 4089b6035b..4cdbb7182b 100644
--- a/projects/Allwinner/linux/linux.arm.conf
+++ b/projects/Allwinner/linux/linux.arm.conf
@@ -1661,7 +1661,7 @@ CONFIG_FIXED_PHY=y
 # CONFIG_MICROSEMI_PHY is not set
 # CONFIG_NATIONAL_PHY is not set
 # CONFIG_QSEMI_PHY is not set
-# CONFIG_REALTEK_PHY is not set
+CONFIG_REALTEK_PHY=y
 # CONFIG_RENESAS_PHY is not set
 # CONFIG_ROCKCHIP_PHY is not set
 # CONFIG_SMSC_PHY is not set
@@ -4065,7 +4065,7 @@ CONFIG_STAGING=y
 # CONFIG_PRISM2_USB is not set
 # CONFIG_COMEDI is not set
 # CONFIG_RTLLIB is not set
-# CONFIG_RTL8723BS is not set
+CONFIG_RTL8723BS=m
 # CONFIG_R8712U is not set
 CONFIG_R8188EU=y
 # CONFIG_VT6656 is not set
```

ToKe79, 2021-05-21:
Just a note - Lakka is currently based on LE 9.2, switch to 10.0 is planned when 10.0 is released by LE team.

ToKe79, 2021-05-22:
@gmello77 can you try image from [here](https://nightly.builds.lakka.tv/members/vudiq/Lakka-LE_master/A64.arm/)?

gmello77, 2021-05-23:
@ToKe79 the image works great. LAN works out of the box. For the audio, I had to add manually in device audio : `default:CARD=allwinnerhdmi` like Gouchi said. If I could choose from the audio menu, instead of typing it would be great.
Thank you, great work btw

ToKe79, 2021-05-23:
@gmello77 thank you for confirming. so as soon as we move to the LE10 base, the support for your device should be as expected. Maybe we can backport some changes from LE10 to current Lakka. For the time being I am closing this issue.

closed 2021-05-23 by ToKe79

klode82, 2022-01-21:
Sorry, I have the same problem of @gmello77, but the image suggested by @ToKe79 is not available. I've tried to search inside Lakka_LE_master, but nothing about A64.arm
Can you suggest me another image to use for ethernet issue?

Thanks.

gouchi, 2022-01-24:
@klode82  You might try this [testing image](https://nightly.builds.lakka.tv/members/vudiq/Lakka-4.0-devel/A64.aarch64/).

Jelte2357, 2023-02-19:
I still have this issue, but the shared testing image links dont work. same device (A64).

gouchi, 2023-02-19:
@Jelte2357 

For [Lakka 4.x](https://nightly.builds.lakka.tv/latest/A64.aarch64/) and [Lakka 5.x](https://nightly.builds.lakka.tv/devbuild_latest/A64.aarch64/) (dev build).

Thank you.

Jelte2357, 2023-02-19:
yes, those dont work. are you saying 3.x could work?

gouchi, 2023-02-19:
You did a make a test  with latest nightly from [Lakka 4.x](https://nightly.builds.lakka.tv/latest/A64.aarch64/)  and [Lakka 5.x ](https://nightly.builds.lakka.tv/devbuild_latest/A64.aarch64/) and it is still not working ?